Klipper3d/klipper #546: COMMAND FOR "FIRMWARE_RESTART" AND "RESTART" PLEASE

CHILLYSMOKES, 2018-08-17:
Someone else may one day want this as well so....

I use OctoPrint's PSU plugin to control my pi's GPIOs using a bash script to automatically switch a set of relays to power on my printer in stages.  I am looking for pi terminal commands that perform the exact same function of FIRMWARE_RESTART and RESTART to insert into the script.  I assume the following:

RESTART = sudo service klipper restart  ??????
FIRMWARE_RESTART = ???????

I am using 2 MCUs and my setup seems to respond better/cleaner when I issue FIRMWARE_RESTART 5 seconds after initial power up and then RESTART 6 seconds after that.  I am just looking to automate this.  Thanks in advance.

AxMod3DPrint, 2018-08-18:
Octoprint has several Plugins to convert GCodes to run system commands or you could set up a GCode Macro in your firmware config, if the PSU control will fire them off.

KevinOConnor, 2018-08-18:
Probably the easiest way to invoke a FIRMWARE_RESTART from the command-line is to run: `echo FIRMWARE_RESTART > /tmp/printer`

-Kevin

oderwat, 2018-08-18:
I don't think you need both `FIRMWARE_RESTART` and `RESTART`. Afaik later one is included with the firmware restart as that one is documented as restart with also reseting the MCU.

CHILLYSMOKES, 2018-08-28:
Point taken. I have removed the last one. Thanks again.