Klipper3d/klipper #547: Support for Variables in printer.cfg

dangerzau, 2018-08-18:
Forgive me if this is the incorrect place to put this request.

I've had a look around and I can't seem to find any refrence to this in the past...

I was simply wondering if there is the availability to utilise variables inside printer.cfg

Reason being - I'm looking for a way to utilise a z-offset not for probing but from the printers "normal" bed height.  Ie if I want to put a dremel spindle into the effector instead of using my delta for printing it would be handy to be able to update a heightoffset variable in side the config by say 5mm (The thickness of the sacrificial board)

Was thinking it would go

$VariableNameHere = SomeValue

Further down in the config..

TowerAPositionEndStop = EndStopValueHere + $VariableNameHere
TowerBPositionEndstop = EndStopValueHere + $VariableNameHere
TowerCPositionEndstop = EndStopValueHere + $VariableNameHere

(Yes I know I haven't used the correct config names for the tower end stops)

if the value is a negative offset this would still work mathematically and if it was a positive value this would work too

If this is already available I'm happy to be schooled on that too 

Cheers

KevinOConnor, 2018-08-18:
Klipper does not support variables in the config file nor does it support expressions in the config file.

For the particular example you've cited, typically one would use the SET_GCODE_OFFSET command to apply a Z offset change.

-Kevin

closed 2018-08-26 by KevinOConnor

JohnEdwa, 2019-01-22:
@KevinOConnor It really needs them, or some other way to do the same, as there are some items that currently require changing values in multiple places hidden in the config files to function properly.

As an example, LCD 'Move Axis' menu currently has hard-coded values. Move 0.1mm is set to move from 0 to 50mm.  Not "move 50 millimeters", but "move *to* 50mm". And it does it really, really slowly.

```
[menu __control __move_01mm __axis_x]
type: input
name: "X:{0:05.1f} "
parameter: toolhead.xpos
input_min: 0
input_max: 50.0
input_step: 0.1
gcode: G1 X{0:.1f}
```

When what it needs to have is `input_min: stepper_x.position_min` , `input_max: stepper_x.position_max`, and `gcode: G1 X{0:.1f} F <stepper_x.jog_speed>` Without this, you need to manually change the values for 10mm, 1mm and 0.1mm for all the axis, and add the changed menu items to your printer.cfg file.

AmyTheCute, 2020-03-23:
I agree to this. for example i want to set my Z offset as a variable since i use it in both my T1 park and activate commands!

ld3300, 2020-03-25:
> I agree to this. for example i want to set my Z offset as a variable since i use it in both my T1 park and activate commands!

For that you can use SET_GCODE_OFFSET at any point in your setup to change the offset.  There's a good example of using it in [sample-multi-extruder.cfg](https://github.com/KevinOConnor/klipper/blob/cff24b0220ffbfe3fe965ea87eb512a898260249/config/sample-multi-extruder.cfg#L26)

ld3300, 2020-03-25:
> @KevinOConnor It really needs them, or some other way to do the same, as there are some items that currently require changing values in multiple places hidden in the config files to function [...]

@JohnEdwa you could pull and test the Menu-V2 commit that is being developed.  Here is an example of what you are describing with the [menu](https://github.com/KevinOConnor/klipper/pull/2344#issuecomment-578430535)

Taudris, 2021-10-02:
To be fair to others, [it is quite easy with SET_TMC_FIELD](https://www.klipper3d.org/G-Codes.html#tmc-stepper-drivers). I even made a macro called `SET_TMC_FIELDS` to make updating multiple motors' parameters easier:

```
[gcode_macro SET_TMC_FIELDS]
gcode:
    {% set steppers = params.STEPPERS|default("stepper_x,stepper_y") %} ;set default steppers here
    {% for stepper in steppers.split(',') %}
      {% if stepper != "" %}
        SET_TMC_FIELD STEPPER={stepper} FIELD={params.FIELD} VALUE={params.VALUE}
      {% endif %}
    {% endfor %}
```

So for example, you could write:

```
SET_TMC_FIELDS STEPPERS=stepper_x,stepper_y,stepper_z1,stepper_z2 FIELD=TBL VALUE=2
SET_TMC_FIELDS FIELD=TOFF VALUE=3 ;only affects the default steppers
;etc
```

My problem with using this approach for real use is what the documentation has to say about `SET_TMC_FIELD`:

> This command is intended for low-level diagnostics and debugging only because changing the fields during run-time can lead to undesired and potentially dangerous behavior of your printer.

I wonder if you could abuse `[include xxxxx.cfg]` for this purpose. Depends on how much validation Klipper does before dropping in the contents of the file.