Klipper3d/klipper #549: "force_move" looking for help with this

bruce356, 2018-08-18:
Hi in printer config I have entered 
[force_move]
enable_force_move: true

I would like my printer to start with a default setting of X-41, y0, z0  before homing so that I can move all the axis' manually to positive  locations (as can be done in Marlin).

I have tried to place 
[force_move]
enable_force_move: true
gcode:
   SET_KINEMATIC_POSITION X=-41 Y=0 Z=0
but this does not work, I have tried several other headings and locations without luck.
Could some one please guide me on how this must be configured.

The gcode functions when entered into octopi terminal but otherwise I just get an error
 Option 'set_kinematic_position' is not valid in section
Thanks - bruce

AxMod3DPrint, 2018-08-19:
Please provide your klippy.log so we can guide you better and work out what mistakes have been made.

bruce356, 2018-08-19:
Thanks for the reply
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2300161/klippy.zip)
regards - bruce

AxMod3DPrint, 2018-08-19:
OK, 2 things. You're using @mcmatrix testing fork for the LCD, which I'm not sure that has been re-based to include `force_move` also the following lines do not belong in the `force_move` section:

```
gcode = 
	SET_KINEMATIC_POSITION X=-41 Y=0 Z=0
```

The `SET_KINEMATIC_POSITION` is an extended gcode which must either be called from a GCode Macro or typed into the terminal and not in the config.

bruce356, 2018-08-19:
Sorry I uploaded the wrong log file but I have corrected this. So you are saying that I need to create a macro to use this command, how would I then load the macro automatically so that XYZ is set at startup of Klipper.
[gcode_macro G82]
gcode: 
	SET_KINEMATIC_POSITION

thanks - bruce