Klipper3d/klipper #552: Wanhao i3 Mini - 'MCU 'mcu' shutdown: Not a valid ADC pin'

CameronLamont, 2018-08-19:
Attempting to setup my Wanhao i3 Mini for use with Klipper.  I've used the Marlin config files as a basis for the pinout to configure Klipper.

I've been able to successfully flash the printer and it does appear to be communicating happily with the RPI3 but I haven't been able to get past the 'not a valid ADC pin' error.

The attached log file shows my incremental fixes to the cfg to make it work - having trailing whitespace and comments after pin names seems to cause grief.  I was putting comments for the original Marlin mappings I was using but eventually took them out.  I was starting to think of writing a tool that parses Marlin into Klipper format (but maybe once I get it running)

I am not super confident that I have correctly mapped the pins - I noticed some of the pins appear in both arrays in pins.py (e.g. PF0 for the Arduino Mega).

Not sure if I still have a syntax issue with my cfg or if I have used the wrong pins.  The links I've been using come from https://www.arduino.cc/en/Hacking/PinMapping2560 and http://forum.arduino.cc/index.php/topic,13779.0.html.  I also looked at the Marlin for the i3 Plus vs the config example for the i3 Plus as a 'primer' to ensure I was mapping them successfully (e.g. pin 61 is a VCC pin on the pinout unless I remap).

The firmware_restart didn't seem to work reliably in some cases until I swapped the restart_method to rpi_usb, not sure if that's best suited here.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2300326/klippy.log)

[printer-wanhao-duplicator-i3-mini.cfg.txt](https://github.com/KevinOConnor/klipper/files/2300328/printer-wanhao-duplicator-i3-mini.cfg.txt)

Let me know if there's any other info I can provide. Thanks in advance

AxMod3DPrint, 2018-08-19:
It looks like the Board is an ATMEGA 2560 as such the pins should be in the RAMPS/MEGA format:

```
[stepper_x]
step_pin: ar22
dir_pin: !ar23
enable_pin: !ar57
step_distance: 0.0125
endstop_pin: ^!ar0
position_endstop: 0
position_max: 120
homing_speed: 30.0
```
The pin numbers should directly translate to Klipper, but with `ar` in front of it.

KevinOConnor, 2018-08-19:
As @AxMod3DPrint indicates, if the only schematic you have uses the arduino pin names then it is easiest is to use those pin identifiers directly in the Klipper config file using the "ar" prefix.  Note, though, that where a sensor pin is used, the prefix is "analog" (eg, the extruder sensor should be "analog13").

-Kevin

CameronLamont, 2018-08-19:
OK thanks guys. I shall re-attempt.

Presumably the sensor pins must be analog? Such as the temp pins?

closed 2018-08-21 by CameronLamont