Klipper3d/klipper #553: Feature Request - Scripting capabilities and host hardware interface libraries

EmbDevices, 2018-08-20:
Scripting capabilities – add possibility to define custom scripts for hardware initialization, communication etc. For example, some hardware devices can be configured, communicate with via custom scripts without need to develop special software support.

Support for host hardware interface libraries. Add possibility to communicate with hardware devices from the host. Porting libraries like [WiringPI](http://wiringpi.com), [WiringOP](https://github.com/zhaolei/WiringOP) etc. to allow high speed interface to devices without need to go through MCU/s.

I tried adding WiringOP to current linux port, got very slow speed.

dragonnn, 2018-08-20:
Most CPU with runs Linux aren't designed to have fast GPIO, so this will never work.

EmbDevices, 2018-08-20:
Before posting such comments what will work or not, you should probably check what speed other people have achieve. For example, RaspberryPi - GPIO Speed Test on a Raspberry Pi (https://swehacker.com/?p=382).
As running single app on [OrangePI One](http://www.orangepi.org/orangepione/) with [WiringOP](https://github.com/zhaolei/WiringOP) I was able to toggle pin at about 1.7 MHz.

Based on the response from other topics, I understand there is some unwillingness about this, but in my opinion this serial connection between the CPU and MCUs, or actually the whole Klipper architecture currently is good enough just for slow speed applications.

KevinOConnor, 2018-08-21:
It's not clear to me what is being requested.  In any case, I suspect you'll need to perform the development work to make it happen.

With respect to issues 133/485, there's no unwillingness.  These types of things require a lot of hard work and (as far as I know) no one is currently interested in doing that work.

-Kevin

theopensourcerer, 2018-08-21:
@EmbDevices The file works fine. Copy and save the raw file into a plain text file on your computer using a simple text editor, not Word or something like that:

https://raw.githubusercontent.com/KevinOConnor/klipper/master/docs/prints/square.stl

If your Slicer still can't open it then you probably have a more fundamental problem. Perhaps try using a different slicer like the excellent Prusa edition of Slic3r, or Ideamaker?

EmbDevices, 2018-08-21:
@theopensourcerer, Thank you about your comment. Yes, the GPL license of this project to some extent is a problem. Many of those hardware companies have horrible licensing policies, they do not comply with GPL, have NDAs about accessing chips software, documentation etc. By the way that is one of the reasons I do not deal with Broadcom and will never purchase device like RaspberryPi. The Asians, Allwinner, Nuvoton, Ingenic etc. are not that bad, software and documentation are available, tools, cheaper etc.

closed 2018-08-28 by KevinOConnor