Klipper3d/klipper #554: Mesh Bed Leveling Save Feature For Manual Probe

ygtc, 2018-08-20:
Hi, i love to see mesh bed calibration feature and i want to ask question or maybe request i can't realize.

I m using mesh bed leveling with manual probe so there is no pnp,capacitive sensor so when i calibrate my bed and create mesh, when i restart firmware, my current mesh is gone.Is there a setting for keep forever like bed tilt? Or we have to generate mesh every time when reboot? 

Keep up to great work! Thank you from now:)

ygtc, 2018-08-20:
#555 
That's what i m talking about:)
@Arksine

derdigge, 2018-08-25:
+1 for mesh safing

lenne0815, 2018-08-29:
That would be great !

KevinOConnor, 2018-10-04:
It should now be possible to save/restore bed mesh points (primarily as of commit b261d31e).

ygtc, 2018-10-04:
yeah i see that. Thanks to this feature i closed issue now

closed 2018-10-04 by ygtc