Klipper3d/klipper #556: TMC2660 support

smark-, 2018-08-20:
Since Klipper already supports the TMC21xx family via SPI, I was wondering if there were any plans to support the TMC2660 drivers as well, since they are featured on the Duet 2 Wifi.

Maglin, 2018-08-21:
They are already supported.  Well, we (the Voron community namely Ampex) got it implemented in the SAM4E Flash to run Klipper on a Duet.  If I'm not mistaken all the available 2660 functionality is there that is available to the Duet.  I believe some stuff wasn't available due to how the Duet is constructed.

KevinOConnor, 2018-08-23:
My understanding is that @FHeilmann (and perhaps others) have been working on support for it.

-Kevin

StefanMeGit, 2018-08-24:
@maglin is Klipper working in the duet board and controlled by the web interface without octoprint? 😳 that’s all wat I am searching 👍🏼

FHeilmann, 2018-08-24:
Klipper working on duet yes-ish (USB doesn't work yet, you need a serial to USB converter, the hardware spis don't work yet)

Web interface without octoprint on because I don't think it makes a lot of sense (yes the hw is there but it goes against the philosophy of klipper imo)

StefanMeGit, 2018-08-24:
That sounds awesome! Is there any documentation to find how to flash ect. I have an DuetWiFi Clone there and I am highly interested to try it out. The board is awesome... would be cool to use Klipper with the Build in Webinterface from the duet.

FHeilmann, 2018-08-24:
Since the Web interface isn't part of reprapfirmware, flashing klipper will result in no Web interface on the board. As mentioned earlier in this thread, tmc2660 support is still under development so I would advise against using it for now.

FHeilmann, 2018-09-11:
https://github.com/KevinOConnor/klipper/pull/621 has been merged, have fun with the TMC2660 :)

closed 2018-09-20 by KevinOConnor