Klipper3d/klipper #557: Timer too close issue

henkyprayoga, 2018-08-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2303654/klippy.log)
It's always happens after SET_PIN PIN=ps_on VALUE=1, G28 then G0 Z500

Thanks!

KevinOConnor, 2018-08-21:
Unfortunately, you've hit an issue with your large Z axis and small Z axis step size.  With the current settings, a Z500 move request requires the host to generate the timing of ~720K steps, and the rpi is unable to do that in its allocated time.  There are a variety of ways to work around this - the easiest is to just issue ten back-to-back Z50 moves instead of one large move.  (It's also possible to tweak the low-level buffer timings if changing the input is not feasible.)

-Kevin

closed 2018-08-30 by KevinOConnor

frankvdh, 2018-12-19:
nigh v,an d

KevinOConnor, 2019-05-05:
FYI, this is now being tracked in issue #1460.

-Kevin