Klipper3d/klipper #558: Question: Mesh Bed Leveling on a delta circular bed

micheleamerica, 2018-08-21:
Hi Kevin

I've got a question related to the Mesh Bed Leveling, both in automatic and manual mode, in a delta printer with a circular bed.
How can I configure it, in relation to the usable bed radius, so that I'm sure that all probed points fall within the usable part of the bed?
From my understanding of the configuration file it would seem that it only supports a square mesh defined by the min_point and max_point corresponding to opposing corners of the square. Is this correct or I'm missing something?

Thanks in advance.

Michele

Arksine, 2018-08-21:
You are correct.   In its current implementation bed_mesh can only generate a rectangular mesh, making it less than desirable for delta printers.   I do plan on investigating alternative ways of building and mapping the mesh to build surface in the future to support a circular or hexagonal mesh.

micheleamerica, 2018-08-21:
Thank you for the update.
Would it not be simpler, as a workaround but maybe as a feature, to be able to designate a specific set of points to be probed? Although I imagine that allowing the probed points to be individually specified would complicate the interpolation algorithm too much.

Arksine, 2018-08-21:
I initially had the user provide points themselves, but there are several formatting requirements necessary to build the mesh.  In its current state the mesh interpolation and lookup algorithms can only accommodate a rectangular mesh, providing a hexagonal grid of points would cause an error.

My conclusion is it was easier on the user to auto-generate them, particularly if they want to probe a large grid.

KevinOConnor, 2018-10-11:
I'm going to close this as I think the original question has been answered (circular beds are not currently supported).  There are other open issues tracking this as a feature request.

-Kevin

closed 2018-10-11 by KevinOConnor