Klipper3d/klipper #564: Invert Z probe?

blaskkaffe, 2018-08-23:
I made a simple Z probe to try the bed leveling feature (will upgrade it later) of an normally open switch i had laying around. The problem is that I cant seem to find a way to make it trigger correctly.

When it is open it says triggered and when it is closed it says open. I cant chose both ^and ! for the pin so that didnt work properly, is there any other way of doing it?

KevinOConnor, 2018-08-23:
One can both invert and pullup - for example: ^!ar3

-Kevin

blaskkaffe, 2018-08-24:
Ah I had tried !^ar18 which gave me an error.

Tried ^! and got it to work! Thanks alot!

Would be great if the documentation mensioned that one way works and the other doesn't

closed 2018-08-26 by blaskkaffe