Klipper3d/klipper #565: Problem Servo Jittering

budiman7x, 2018-08-24:
hello
i have a problem using servo and microswicth for z probe

i am currenly using HXT900 or HK15148B, both of them trying in marlin firmware is steady, but on klipper get jittering and getting worse if any motor 3D printer on moving
i am trying to get external powersupply still get bad jittering

this is my setting

`[servo my_servo]
pin: ar19
maximum_servo_angle: 200
minimum_pulse_width: 0.0006
maximum_pulse_width: 0.0026

[probe]
pin: ^ar18
z_offset: 12.80
activate_gcode:
   	SET_SERVO SERVO=my_servo ANGLE=10
    	G4 P200
deactivate_gcode:
    	SET_SERVO SERVO=my_servo ANGLE=90
	G4 P100
	

[bed_tilt]
horizontal_move_z: 25
#x_adjust:
#y_adjust:
points:
    10,65
    85,65
    165,65
    165,145
    85,145
    10,145
    10,210
    85,210
    165,210
`
any recommendation for my issue? thank you

BKLronin, 2018-08-27:
This happens on marlin as well even with filters and dedicated 5V power. Switching off the servo after leveling /usage would be the easiest solution I guess

budiman7x, 2018-08-27:
I got the solution, but still not perfect
Using default value from config

maximum_servo_angle: 180
minimum_pulse_width: 0.001
maximum_pulse_width: 0.002

Very minimum jittering in result, thank you

closed 2018-09-06 by KevinOConnor