Klipper3d/klipper #567: Z_TILT_ADJUST fails in Voron 2.0

eldonpinheiro, 2018-08-24:
I spent several hours trying to understand the operation of this command.

I have a Voron 2.0 with 4 steppers moving the 4 corners of the gantry.

After the G32 command, the procedure is initialized and after finishing, it gets some systematic errors, always leaving 2 opposite sides higher than the others.

I have attached the log for understanding.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2318808/klippy.zip)

derdigge, 2018-08-24:
@eldonpinheiro 
Please add your config.

KevinOConnor, 2018-08-24:
@derdigge - the config is in the log file.  We ask that issues contain the log file (as described at https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md ) - it's not necessary for a separate config file to be attached.  If you'd like to take a look at this issue (thanks!) then download the log and run it through `scripts/logextract.py` as described in https://github.com/KevinOConnor/klipper/blob/master/docs/Debugging.md#extracting-information-from-the-klippylog-file

derdigge, 2018-08-24:
@KevinOConnor Thanks, indeed.

KevinOConnor, 2018-08-25:
@eldonpinheiro - make sure that all the z_tilt probe points are in bounds of the bed (the coordinates are coordinates of the nozzle, but the probe must be above the bed at that nozzle coordinate).  Make sure the coordinates of the z_steppers are correct (again, this should be the coordinate of the nozzle, could the nozzle move to directly above each stepper motor).  Finally, avoid using 4 probe points - use either three points (make sure they form a triangle), or use 9+ probe points.  The use of four points increases the risk the code perceives a "potato chip" signal.

-Kevin

eldonpinheiro, 2018-08-25:
Hi Kevin,

I'm using this layout to set coordinates of steppers.

```
 ----------------
 |Z1          Z2|
 |  ---------   |
 |  |       |   |
 |  |       |   |
 |  x--------   |
 |Z           Z3|
 ----------------
```
I used 25 probe points this time. It has improved a bit but still remains with a systematic error.
I put the values in an array in excel and did a plot, look how it was:

[image]

Attached log:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2320767/klippy.zip)

KevinOConnor, 2018-08-25:
Okay, I think I understand.

The current z_tilt code is based on the premise that the bed is rigid and flat, but may be tilted.  The z_tilt code finds the tilt and alters the steppers to undo that tilt.

If I understand you correctly, the bed isn't just tilted it is slightly deformed.  I'm guessing you want to move the steppers to attempt to undo that deformation.  For example, by lowering steppers Z1 and Z3 while keeping steppers Z and Z2 stationary.  Alas, the z_tilt code does not support that.

I could see where that feature would be useful.  I suspect it could be made to work if someone knows of a math formula expressing how much to move each stepper based on the probe results.  It's not clear to me how to do that.  (I see that Z1 and Z3 need to be lowered wrt Z/Z2, but by how much?)

Maybe @Maglin has some ideas?

-Kevin

derdigge, 2018-08-27:
@eldonpinheiro 

You can try the following. I did same with 3point leadscrew on my Hypercube.
Adjust yout Point section like this (assuming you have 300x300) on your voron2.

`[z_tilt]
z_positions = 
	-45,-35
	-45,390
	405,390
	405,-35
points = 
	50,50
	50,50
	50,250
	50,250
	250,250
	250,250
	250,50
	250,50
speed = 200
horizontal_move_z = 10`

The rest should(could) be done with the meshleveling. You call BED_MESH_CALIBRATE but i cant see a config of mesh leveling.

derdigge, 2018-09-12:
Thanks Kevin!

closed 2018-09-20 by KevinOConnor