Klipper3d/klipper #568: Working LCD Menu - Basic Pre-Configured Example

cfelicio, 2018-08-24:
I have been away from Klipper for a while, but got a bit of time available, and wow, a lot of things have changed in 1 year! Amazing work!!!

I know the LCD menu is still in progress, but I saw the example menu cfg, and I'm wondering if we could come up with something to allow users to have basic menu functionality, without having to put too much thought into it.

I looked at the menu for Repetier and Marlin, and some common  basic features among them:

- Preheat Extruder / Bed
- Cooldown
- Home Axis (All and Individually)
- Move Axis (Marlin and Repetier do this differently, but basically you have options to move fast and slow)
- Disable Steppers

Does this make sense? It seems like the configurations might be slightly different depending on what type of LCD people have, so perhaps we can break down the examples to the most common types of LCDs as well, so everyone can have a base to start of?

Comments? Should I start looking into this? Any other menu entry suggestions to add to the basic list? :-)

mcmatrix, 2018-08-24:
It’s not wip anymore. It’s in main code now.
To have the menu you just have to add few more attributed into display section.
encoder_pins: ^!ar31, ^!ar33
click_pin: ^!ar35
Or whatever are your pins for these.

Klipper has a builtin default menu structure. You don't have to add anything else but just right pins and you have a working menu.

cfelicio, 2018-08-24:
oh, that's excellent news! I will pull the latest code tonight and try it out! Thanks so much! :)

closed 2018-08-24 by cfelicio