Klipper3d/klipper #569: Finding pins for Melzi (Wanhao i3)

cfelicio, 2018-08-24:
So, now that we have a working menu for Klipper, I'm wondering how to find the proper pins for the wanhao i3:

There is already basic LCD support on the example configuration:

[display]
lcd_type: st7920
cs_pin: PC1
sclk_pin: PD3
sid_pin: PC0

Following suggestions from other members, I had a look here:

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/pins_SANGUINOLOLU_11.h

Looks like the corresponding pins:

#if ENABLED(IS_MELZI)
#define LCD_PINS_RS     30   // CS chip select /SS chip slave select
#define LCD_PINS_ENABLE 29   // SID (MOSI)
#define LCD_PINS_D4 17 // SCK (CLK) clock

Then I also found these:

#define BTN_EN1 11
#define BTN_EN2 10

But how do I translate these to the format being followed on the config (PC1, PD3, etc)? Any ideas? :)

Thanks!

KevinOConnor, 2018-08-25:
I've added a new section to the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-convert-a-marlin-pin-number-to-a-klipper-pin-name

Hopefully that helps.
-Kevin

cfelicio, 2018-08-25:
Excellent guide, thanks so much! I was able to find the pins and get it working. I sent a PR to add it to the example cfg file. Cheers! :-)

closed 2018-08-25 by cfelicio