Klipper3d/klipper #572: Allways let motors on?

derdigge, 2018-08-25:
Hello.

How to set stay motors on? If Z falls down on motors off. I went throug example configs and found nothing regarding to this.

Thanks.

StefanMeGit, 2018-08-25:
# Idle timeout. An idle timeout is automatically enabled - add an
# explicit idle_timeout config section to change the default settings.
#[idle_timeout]
#gcode:
#   A list of G-Code commands (one per line) to execute on an idle
#   timeout. The default is to run "M84".
#timeout: 600
#   Idle time (in seconds) to wait before running the above G-Code
#   commands. The default is 600 seconds.

From the example-extras

derdigge, 2018-08-25:
@StefanMeGit 

Yes saw this but i dont wand to turn them off, not increasing a timeout.

[idle_timeout]
gcode:

so this will do nothing when timeout reached?
Or set timeout to 0?

StefanMeGit, 2018-08-25:
Give it a try... set the timer to 10 seconds and leave the code section empty. Should work.

derdigge, 2018-08-26:
yes it did. Great. Thank you

closed 2018-08-26 by derdigge