Klipper3d/klipper #576: Heater bed temperature run away

roverrobot, 2018-08-26:
Printing the square.stl for pressure advance tuning using ABS with hot bed set to 100 degree C.
Printer: Anet A8
Klipper: git master
OctoPrint: 1.3.9
Slicer: Cura 3.4.1
Problem: firmware stops with error "Heater heater_bed not heating at expected rate."

I wonder what causes this. The happens for both PID and watermark controls (watermark following PID in the attached log). The PID values are taken from tuned values, also in the attached log.The drift can be seen from the attached temperature plots.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2321999/klippy.log)
[image]
[image]

blaskkaffe, 2018-08-27:
Happened to me too yesterday, upgraded to the latest master and suddenly the extruder didnt heat at expected rate, added the lowest extrusion temperature to settings which fixed it.

KevinOConnor, 2018-08-28:
@roverrobot - yes, it appears the power supply is not powerful enough to maintain the bed heater at 100 degrees.  The log indicates that even without other components drawing power, the bed heater needs to be run at 100% to even maintain 100 degrees.

-Kevin

closed 2018-09-03 by KevinOConnor

adelalbuaijan, 2018-09-28:
Gus I am having the same issue with CR-10s once the Bed reaches its target which is 65 and the Extruder reaches 210 and the Printer starts printing, Bed Heat starts to go low and I recorded that after 20 degrees I get Thermal Runaway !! I don't know how to fix this issue, please pease help.