Klipper3d/klipper #577: Internal error on command:"G28"

MGX3D, 2018-08-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2322387/klippy.log)

Running into an odd error with this combination:
- Trinamic TMC2130 (Einsy Board)
- 0.9 degree steppers
- 2mm/turn ballscrew (THK KR26)

I agree, not the typical 3d printer setup :) but it can run at 15mm/s max with micro-stepping up to 8x, that is __0.625 microns/step__, but any higher microstepping will result in an *"Internal error"* while homing the Y axis.  If my calculations are correct, at 32x microstepping for example it would need .15 microns/step (160Khz pulse at 25mm/s), so not sure if this is achievable in Klipper.

An additional artifact (likely unrelated) is that if I increase the current above 0.5amps, the motor stalls even at 15mm/s. 1.8 degree motors with a faster ballscrew (6mm/turn) seems to run fine up to 50mm/s (axis X in this config).

__Relevant section from the log__:
> Read 276807.113226: 'G28 Y\n'
gcode state: absolutecoord=True absoluteextrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=25.0
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 169, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 69, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 489, in cmd_G28
    homing_state.home_axes(axes)
  File "/home/pi/klipper/klippy/homing.py", line 114, in home_axes
    self.toolhead.get_kinematics().home(self)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 100, in home
    self._home_axis(homing_state, axis, self.rails[axis])
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 77, in _home_axis
    homing_state.home(coord, homepos, rail.get_endstops(), homing_speed)
  File "/home/pi/klipper/klippy/homing.py", line 104, in home
    self._homing_move(movepos, endstops, speed)
  File "/home/pi/klipper/klippy/homing.py", line 64, in _homing_move
    mcu_endstop.home_wait(move_end_print_time)
  File "/home/pi/klipper/klippy/mcu.py", line 191, in home_wait
    while self._check_busy(eventtime, home_end_time):
  File "/home/pi/klipper/klippy/mcu.py", line 219, in _check_busy
    raise error("MCU is shutdown")
error: MCU is shutdown
Internal error on command:"G28"
Stats 276808.8: gcodein=59 mcu: mcu_awake=0.021 mcu_task_avg=0.000172 mcu_task_stddev=0.000245 bytes_write=3497 bytes_read=5873 bytes_retransmit=9 bytes_invalid=6 send_seq=169 receive_seq=169 retransmit_seq=2 srtt=0.005 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15999982 heater_bed: target=0 temp=24.1 pwm=0.000 print_time=24.318 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.2 pwm=0.000
Stats 276809.8: gcodein=64 mcu: mcu_awake=0.011 mcu_task_avg=0.000110 mcu_task_stddev=0.000132 bytes_write=3503 bytes_read=6002 bytes_retransmit=9 bytes_invalid=6 send_seq=170 receive_seq=170 retransmit_seq=2 srtt=0.005 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15999880 heater_bed: target=0 temp=23.9 pwm=0.000 print_time=24.318 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000

KevinOConnor, 2018-08-28:
> I agree, not the typical 3d printer setup :) but it can run at 15mm/s max with micro-stepping up to 8x, that is 0.625 microns/step, but any higher microstepping will result in an "Internal error" while homing the Y axis.

The root error is a "Rescheduled timer in the past" which is an indicator that the mcu has been asked to step too high.  A homing move requires more overhead than regular moves (the mcu has to check the endstop in addition to stepping).  So, it may be possible to get to faster run-time speeds if you home with a slower homing_speed.

-Kevin

MGX3D, 2018-09-02:
I'm also getting scheduler errors at 60KHz during G0 moves (300mm/s at 32x microstepping, 5 microns/step) on a corexy. On the same Einsy board. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2343735/klippy.log)

[image]

KevinOConnor, 2018-09-03:
Right - on corexy an X100 move requires both steppers to step in concert (both the X+Y and X-Y steppers must step to move the toolhead).  So, at 300mm/s that's 2*60K == 120K, which is greater than the atmega2560 benchmark.

-Kevin

MGX3D, 2018-09-03:
Oh, I think I misread the 150K/100K for AVR16MHz (I assumed 100K is max) from here: https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md

I'm going to give a LPC1769 a try. Thanks!

closed 2018-09-03 by MGX3D