Klipper3d/klipper #578: Internal error on command:"M84"

BKLronin, 2018-08-27:
There seems to be a an error in step compression on the polar branch, log file said this.

Printer is heating up and homing axis but when starting to print Error occurs.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2323749/klippy.log)

BKLronin, 2018-08-27:
#174 Seems to be the same issue. Trying to do the same.

BKLronin, 2018-08-27:
With the modified homing delay value I get:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2324808/klippy.log)

KevinOConnor, 2018-08-28:
This is due to the inability of the demo code to properly handle moves around position 0,0.  To workaround this, try changing the homing_override in your config to something like:
```
[homing_override]
axes = xyz
gcode =
	G28 X
        G1 X20 Y0 ; Move away from origin
set_position_z = 0
```

-Kevin

BKLronin, 2018-08-28:
It seems to work!  :)

Printed a nice rectangle and I´m impressed how your democode worked at first try. Apart from my failures. 

Improving the machine a bit now..

*Happy*

KevinOConnor, 2018-09-06:
Any further progress on testing?

BKLronin, 2018-09-06:
I´m currently collecting some short video sequences, while printing benchys to determine the mechanical upgrades needed to ramp up the speed even further. 

But in general I can print mechanically accurate when not in the center. More detailed testing with protocol will follow when the machine has been upgraded.

BKLronin, 2018-09-06:
The Leveling is a bit annoying, once it is adjusted it works but leveling with probe and the suggested three point independent stepper adjustment for horizontality would be a breeze. Not possible on the hardware yetbut I think that might be necessary for fire and forget prints.

KevinOConnor, 2018-09-25:
Okay - I'm going to close this issue for now.  If you have further feedback, let me know.

-Kevin

closed 2018-09-25 by KevinOConnor

BKLronin, 2018-09-25:
I´m returning to this project periodically as time frees up. 

The hardware upgrades are still in the waiting line. Software is usable my odroid just crashed one time and some overheating probs. That means the hardware is a bit behind the possiblitys.

Uploaded the promised video for visual inspection of the performance. On the slow side at the moment because of the missing part cooling fan. Lots of headroom with the cpu at 5%. 

Weill continue soon, in the meanwhile 
[](https://youtu.be/y0ZbRfyshag)

BKLronin, 2018-09-26:
Yes, I think its ok for general use and experimentation. 

I´m going to upload my file as soon as I´m at the odroid, maybe with some comments for the calculation. Might take a few days.

Thomas

BKLronin, 2018-10-09:
Hey Kevin, I´ve attached the curently working printer.cfg that I´m using. 
[printer-raise_uav-Hullk-polar.zip](https://github.com/KevinOConnor/klipper/files/2460545/printer-raise_uav-Hullk-polar.zip)

@KevinOConnor

KevinOConnor, 2018-12-20:
FYI, I plan to add the polar kinematics to the master branch and mark it as "experimental" - see PR #1023.

-Kevin

BKLronin, 2019-06-10:
I had a look at the continous rotation task again. The newer approach made it easier for me.
Can´t test at the oemnt but could it be possible to get away with this additional line?

static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m
                                  , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    // XXX - handle x==y==0
    // XXX - handle angle wrapping
    int angle =  atan2(c.y, c.x); 
    if angle > 360 then angle -360; 
    
    return angle;
}

KevinOConnor, 2019-06-10:
Sort of.  The atan2() function returns a value between -pi and +pi, but you want the result of polar_stepper_angle_calc_position() to return a continuous value (no large jumps in return value).

Something like:
```
static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m
                                  , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    // XXX - handle x==y==0
    double angle = atan2(c.y, c.x);
    if (angle - sk->commanded_pos >= 1.50 * M_PI)
        angle -= 2. * M_PI;
    else if (angle - sk->commanded_pos <= -1.50 * M_PI)
        angle += 2. * M_PI;
    return angle;
}
```

But the above wouldn't work for long moves (that span more than two quadrants) or if the table rotated more than once during a print.

-Kevin

BKLronin, 2019-06-10:
Hmm interesting, I forgot it is -180 + 180 so in the meantime I wrote this.

`static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m
                                  , double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    // XXX - handle x==y==0
    // XXX - handle angle wrapping
    // if rotaion  coordinate = 360°
    //float pi2 =  6,2831853071795864 
    float pi =  3.14159265359;
    float angle =  atan2f(c.y, c.x); 
    float angleDiv = 0;
    if (angle > pi)
      angleDiv = angle - pi;
      angleNeg = - pi + angleDiv;
    else 
      anglecont = angle;
    
    return angleCont;
}`

I assume the path planner for continous extrusion gets into trouble and can´t calculate a move? 

Taking some time to study your code again...

BKLronin, 2019-06-10:
Can we move the bed in relative commands somehow? That would be ideal.

BKLronin, 2019-06-10:
Maybe like this for x and y  (pseudo code) and then switch to relative commands via gcode?

`
static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m
                                  , double move_time)
{
    struct coord s = coord->start_pos
    struct coord c = move_get_coord(m, move_time);
    // XXX - handle x==y==0
    double angleStart = atan2(s.y, s.x);
    double angle = atan2(c.y, c.x);
    double angleDif = angle - angleStart
`

BKLronin, 2019-06-11:
My thoughts: if the distance is longer than a 360 degree turn in conjunction wit r then make it even a more negative or more positive value instead of jumping to the opposite sign. Something like that, not sure where to get a vlaue to measure if it is a huge step without comparing with last step. 

```
static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m
                                  , double move_time)
{

    struct coord c = move_get_coord(m, move_time);
    double d = move_get_distance(m, move_time);
    // XXX - handle x==y==0
    double angleIn = atan2(c.y, c.x);
    if (d >= >Distance for a 360 degree turn, dependent of radius<) // Unit?
      if (angleIn < 0);
      angle =  AngleIn + (2 * M_PI)

      if (angleIn > 0);
      angle =  AngleIn - (2 * M_PI)      

    else
      angle = angleIn;

    return angle

```

BKLronin, 2019-06-14:
@KevinOConnor 
I tried a few different strategies with the use of the the batch debug. Printing a circle works with your last code snippet to 540 degrees. Thanks for that. This is quiet confusing to me as it should reset at exakt 2 * pi without incrementing the multiplier.

Angle: 9.424778 Count: 1 Start: 8.865184
Angle: 3.141594 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184
Angle: 9.424778 Count: 1 Start: 8.865184
Angle: 3.141594 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184
Angle: 9.424778 Count: 1 Start: 8.865184
Angle: 3.141594 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184
Angle: 9.424778 Count: 1 Start: 8.865184
Angle: 3.141594 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184
Angle: 3.141593 Count: 0 Start: 8.865184

Thats the code where it flips over and hangs at pi. I tried to catch this difference as well but it didn't worked. The idea was to count how often a full rotation has been achieved and then add that value constantly to the continous rotation. I had it working somehow but the angle would not have been continued smoothly.

`int count ;
double D_PI = M_PI * 2;

static double
polar_stepper_radius_calc_position(struct stepper_kinematics *sk, struct move *m, double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    double radius =  sqrt(c.x * c.x + c.y * c.y); 
    //printf ("Radius: %f\n",radius);	
    return radius;
}

static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m, double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    //double d = move_get_distance(m, move_time);
    //printf ("Gmove_get_coord%f\n",c);
    //printf ("Get_Distance%f\n",d);

    // XXX - handle x==y==0
    
    
    double angleStart = sk->commanded_pos;
    //printf ("\rLast_Pos: %f\n",angleStart);	
    double angle = atan2(c.y, c.x);
    //printf ("\rAngleIn: %f",angle);	
    
    if (angle - angleStart >= 1.5 * M_PI )
        {angle -= D_PI;
	//++count;
	}
    else if (angle - angleStart <= -1.5 * M_PI)
        {angle += D_PI;
	//--count;
	}	

    if (angle >= D_PI)	 
	count = angle / D_PI;	
	
    else
 	count = 0;

    

    printf ("\rAngle: %f Count: %d Start: %f\n",angle, count, angleStart);
    //fflush(stdout);
    //printf ("\rCount: %i",count);
    
    return angle;`     

The most important question is: why does it rotate more than 360 degrees?

BKLronin, 2019-06-15:
Math ain right yet and maybe logic as well but thats how imagine it could work.
```
//
// Copyright (C) 2018  Kevin O'Connor <kevin@koconnor.net>
//
// This file may be distributed under the terms of the GNU GPLv3 license.

#include <math.h>      // sqrt
#include <stdlib.h>    // malloc
#include <stdio.h>
#include <string.h>    // memset
#include "compiler.h"  // __visible
#include "itersolve.h" // struct stepper_kinematics

//#define M_PI acos(-1.0)
double count ;
double D_PI = M_PI * 2;

static double
polar_stepper_radius_calc_position(struct stepper_kinematics *sk, struct move *m, double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    double radius =  sqrt(c.x * c.x + c.y * c.y); 
    //printf ("Radius: %f\n",radius);	
    return radius;
}

static double
polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m, double move_time)
{
    struct coord c = move_get_coord(m, move_time);
    //double d = move_get_distance(m, move_time);
    //printf ("Gmove_get_coord%f\n",c);
    //printf ("Get_Distance%f\n",d);

    // XXX - handle x==y==0
    
    
    double angleStart = sk->commanded_pos;		//Get the last command in polar coordinates this class has produced.
    //printf ("\rLast_Pos: %f\n",angleStart);	
    double angle = atan2(c.y, c.x);
    //printf ("\rAngleIn: %f",angle);	
 
    if (angle >= D_PI * 1.49999) 			//Test if we have already rotated the bed past a value close to 3* PI
    	count = (angle / D_PI);	 			//If so calculate the amount of rotaions since we passed +180°
    
    if (angle - angleStart >= 1.5 * M_PI ) 		//Check if the difference to the last command exceeds the treshold of 1,5 * PI
	{
	if (count <1) 					//In case there was no full rotation in one one direction        
		angle -= D_PI; 				//Add diference relative to 180
	else if (count > 1 && angleStart >= 2 * D_PI)	//If we have done a full rotation and the last position as well
		angle -= D_PI * count;			//Extend the rotation by the count multiplier so we don't get another jump after D_PI has been depleted
	//++count;
	}
    else if (angle - angleStart <= -1.5 * M_PI) //Check if the difference to the last command exceeds the treshold of -1,5 * PI
        //{angle += D_PI;
	{
	if (count <1)        
		angle += D_PI;
	else if (count > 1 && angleStart >= -2* D_PI)
		angle += (D_PI * count);
	//++count;
	}
	//--count;
		
   
		
	
    //else
 	//count = 0;

    

    printf ("\rAngle: %f Count: %f Start: %f\n",angle, count, angleStart);
    //fflush(stdout);
    //printf ("\rCount: %i",count);
    
    return angle;
}

struct stepper_kinematics *__visible
polar_stepper_alloc(char type)
{
    struct stepper_kinematics *sk = malloc(sizeof(*sk));
    memset(sk, 0, sizeof(*sk));
    if (type == 'r')
        sk->calc_position = polar_stepper_radius_calc_position;
    else if (type == 'a')
        sk->calc_position = polar_stepper_angle_calc_position;
    return sk;
}
```

BKLronin, 2019-06-16:
Next iteration, as commanded_pos returns something that is further back in time and not exactly the last linear angle, I came to the conclusion that a list is necessary to save the value from the last cycle so it is possible to calculate a more precise threshold. Without that the modifier adds to soon to the angle and causes a step that is too big. 

That "step trigger" needs to be iterated and the the test array replaced with a list. List and memory stuff is more difficult in C than in python, so I have to get into that first. 

[kin_polar.zip](https://github.com/KevinOConnor/klipper/files/3294070/kin_polar.zip)

BKLronin, 2019-06-16:
Working, but CPU on Raspi 1 maxed out :D

KevinOConnor, 2019-06-17:
As a high-level comment, we don't really want to add any complexity to polar_stepper_angle_calc_position() as that code is the "fast path" - it's called many many times during the "iterative solver" mechanism.  Instead, the preferred option would be to break up the moves before calling the iterative solver (eg, in polar.py:move() ).  That code could detect if a move spans quadrants and split moves so that no single move spanned a quadrant.  Once a move is isolated to a single quadrant, the code could call stepper.set_commanded_pos() to set the appropriate angle for that quadrant.

-Kevin

BKLronin, 2019-06-17:
Thanks Kevin, much appreciated.  

I understand, I was able to identify the problem yesterday and what seemed like slowing down by low ressources was more of a small error in my code. At least it rotates multiple time even if it is pretty much a static construct. Downside is that it accumaltes a siginificant error  caused by the modification multiplier trigger that steps up MP_I per rotation. I´ve attached the current working code. 

However, I might be able to implement it the way you suggested. 

Detecting a crossing would be possible without polar conversion, thats good. I still have to do the continous rotation step up mechanism in the C file as I can´t muliply the cartesian space which would result in disortions. I don´t want to convert into polar at this point (py) already because it would be either done twice or slower if I just pass the kinematic_polar.c.

Let me know if it would help to just put the detection into the polar.py and leave the continous stuff in the c file. 

[kin_polar.zip](https://github.com/KevinOConnor/klipper/files/3298023/kin_polar.zip)

KevinOConnor, 2019-06-17:
It should be possible to implement continuous rotation without any changes to kin_polar.c (that is, with polar_stepper_angle_calc_position() just returning the simple atan2() calculation).  Once every move is contained within a single quadrant, one should be able to do something like:
```
    if move.start_pos[0] < 0. and move.end_pos[1] < 0.:
         # Fixup move in quadrant -x/+y
         stepper_bed = self.steppers[0]
         cp = stepper_bed.get_commanded_pos()
         if cp > 0.:
               stepper_bed.set_commanded_pos(cp - 2. * math.pi)
    elif move.start_pos[0] < 0. and move.end_pos[1] > 0.:
         # Fixup move in quadrant -x/+y ...
```

FYI, it's not convenient for me to look at code in a zip file - if you're looking for me to review, I suggest putting it on a branch somewhere on github.

-Kevin

BKLronin, 2019-06-18:
Ah I think I got it now. The reason why the iterative solver returned the error wasn´t the jump itself, it is caused by a move that contains a jump. As it calculates a move or position within a high and low value a jump in that causes him to correctly doubt the sanity of the incoming move.

So if we feed him moves which don´t contain jumps it will solve, regardlessly if the the next move is negative. So it´s more linear and doesn´t require stepwise addition of the multiplier :)

Willd do the next time, wasn´t shure how to submit code correctly.

BKLronin, 2019-06-20:
Hmm, he quadrans don't really help. atan2 already returns he correct ones and I can think of a procedure that doesn produce a large step that fails on the stepcompress.c check.

There always needs to be an upstepping factor in order to rotate he virtual space so i doesn´t  encouner a 360 difference (-180+180 or   0 360). The mechanical counerpart of this is essentiay a crank shaft but we don't have kinetic force to get past the amplitude points. 

```
#quadrant3 = x < 0. and y < 0.
        if endMoveX < 0. and endMoveY < 0.:
            stepper_bed = self.steppers[0]
            cp = stepper_bed.get_commanded_position()
            print("cp_quadrant3", cp)

            if cp < 0.:
                stepper_bed.set_commanded_position(2 *math.pi -cp)
                check = stepper_bed.get_commanded_position()
                print("cp_quadrant3_adjusted", check)

```
Essentially the same problem happens I had in the kinematics.c and that is catching the end of the positive move radiant in time before it turns negative. Somehow adding to the alreaddy negative value seems to late and stepcompress cmplains about the huge step.
 
Don´t really know what to do, thought about countless ways to achieve it, the best would be a polar slicer of course. 

There must be some advanced awesome math thing that projects a static repeating move into infinite rotations :D

KevinOConnor, 2019-06-20:
In order to use the quadrant trick, the moves would first need to be split so that no move spanned multiple quadrants.  Admittedly, that's not easy.

If you just want to get something working - I think the below may work (totally untested).

-Kevin

```diff
diff --git a/klippy/chelper/kin_polar.c b/klippy/chelper/kin_polar.c
index cc1f2eab..0a6ecf9e 100644
--- a/klippy/chelper/kin_polar.c
+++ b/klippy/chelper/kin_polar.c
@@ -24,8 +24,12 @@ polar_stepper_angle_calc_position(struct stepper_kinematics *sk, struct move *m
 {
     struct coord c = move_get_coord(m, move_time);
     // XXX - handle x==y==0
-    // XXX - handle angle wrapping
-    return atan2(c.y, c.x);
+    double angle = atan2(c.y, c.x);
+    if (angle - sk->commanded_pos >= 1.50 * M_PI)
+        angle -= 2. * M_PI;
+    else if (angle - sk->commanded_pos <= -1.50 * M_PI)
+        angle += 2. * M_PI;
+    return angle;
 }
 
 struct stepper_kinematics * __visible
diff --git a/klippy/kinematics/polar.py b/klippy/kinematics/polar.py
index 2930152c..b40be954 100644
--- a/klippy/kinematics/polar.py
+++ b/klippy/kinematics/polar.py
@@ -122,8 +122,14 @@ class PolarKinematics:
         axes_d = move.axes_d
         cmove = move.cmove
         if axes_d[0] or axes_d[1]:
-            self.steppers[0].step_itersolve(cmove)
             self.rails[0].step_itersolve(cmove)
+            stepper_bed = self.steppers[0]
+            stepper_bed.step_itersolve(cmove)
+            angle = stepper_bed.get_commanded_position()
+            if angle < -math.pi:
+                stepper_bed.set_commanded_position(angle + 2. * math.pi)
+            elif angle > math.pi:
+                stepper_bed.set_commanded_position(angle - 2. * math.pi)
         if axes_d[2]:
             self.rails[1].step_itersolve(cmove)
```

BKLronin, 2019-06-20:
Thanks Kevin,

Thats essentially what we have already. It worked to an angle to 900 Degrees and more but had to be stepped up finitely as the rotation progresses. It is finite as we have the constant of 2 * pi in the fixed move.

 I actually don´t think that a quadrant trick could work as even if it is segmented to the actual quadrant the next move would still be out of range. Maybe I´m missing something but that should at least trigger a full 360 turn in the opposite direction before continuing. So the Printer needs  to break and and go full reverse and all that. 

I did segmentation of gcode before that handled a certain length but as you said I doubt I can implement that the right way.

KevinOConnor, 2019-06-20:
> Thats essentially what we have already. It worked to an angle to 900 Degrees and more but had to be stepped up finitely as the rotation progresses.

Do you have a test case that demonstrates the failure?  It should scale to any number of rotations.

-Kevin

BKLronin, 2019-06-20:
Well, IT DOES!!!!

I have to apologize I just read the code and just found the minor change with the cmove and assumed its basically the same with a different implementation.

You have no idea how releiving this is and I have no idea why it is working :D

THIS IS AWESOME, have to try it ASAP on the printer.

KevinOConnor, 2019-06-20:
Ah - the code had an error in it (it should not use 1.5 * M_PI).  I put together a work-polar-20190620 branch with the latest version of the code above:

cd ~/klipper ; git fetch ; git checkout origin/work-polar-20190620 ; sudo service klipper restart

The above also adds some multiple rotation test cases, which now pass.

> I have no idea why it is working

The iterative solver produces steps relative to its "current" position.  By calling set_commanded_position(), we reset what it thinks is its "current" position.  No single linear move can rotate the bed more than pi radians.  The change to polar.py normalizes the angle so it doesn't "wind up" on future moves.

-Kevin

BKLronin, 2019-06-20:
Ahh I think I got it, essentially :)

I hope I can test this on the machine the next days somehow. Needs a video for sure. Thank you a lot!

KevinOConnor, 2019-06-27:
Did that polar kinematics change work out?

-Kevin

BKLronin, 2019-06-28:
I had a few things that queued up while I spent time with the polar code. So it got a bit delayed but I may be able to do dry run today. :)

BKLronin, 2019-06-28:
Looking good. Continous is working as intended. Tried at 40mm/s @800% :)

Additionally I also tried to print a solid disk object that covers the 0,0 center and as it got close to the center it threw a "Rescheduled timer in the past" error. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3339390/klippy.log)

KevinOConnor, 2019-06-28:
Great.  I pushed the continuous rotation change to the main branch (commit e5f74a64).

Yeah, moves around the origin are likely to result in that type of error.  If one moves with speed near the origin it results in an extremely fast bed rotation - faster than the stepper motors (and even the micro-controller) can achieve.

In order to make moves near the origin work, I think the code would need to split up the moves so that it can slow down the parts of a move that are near the origin.  I think it would mostly work as long as the head isn't travelling fast.

-Kevin

BKLronin, 2019-06-29:
Ok, seems solid, dry printed two parts without errors.

I knew that this is still something that has to be handled. Just wanted to feedback some data directly from the printer. I tried it with 40mm/s and I received a G1 error, that should mean I made it all the way to the zero crossing. 

Splitting up sounds good I don´t have any solid ideas for the center crossing yet.

Thought about: 
- Exclusion zone that produces a hole in the print. Better doing this in the slicer or model itself.
- Pause the X axis and extrusion until the bed has fully rotated slowly with a constant not direct feedback from the back.
- A general feedrate modifier that lowers the speed towards the center so its pretty much equal through the print. Mapped to the radius given in the config.

Sounds like another deep dive  progamming adventure to me :D I might continue working on the fail detection first as I find the polar kin pretty much usable now for hull printing. 

If you got something I can test it pretty fast as the printer is up and running at the moment.

KevinOConnor, 2019-06-30:
FWIW, my thinking is that handling moves near the origin would provide 99% of the necessary support.  That is, if one could successfully run `G1 X100 Y1` followed by `G1 X-100 Y1 F6000`.  To implement that, I think the code could split up the move - more splits for the move segments closer to 0,0 and slower speed on those move segments near 0,0.

That wouldn't handle the case where a move goes directly through the origin (0,0), but I think support for that would be a relatively simple addition.  For the corner case where a move goes exactly through the origin, the move segments nearest the origin could be altered to move to a position just near the origin instead.  That is, we can add an exception so that moves exactly through the origin are treated as moves very near (eg, a quarter of one step distance) the origin.

-Kevin

sanpedrox2, 2019-06-30:
Hi guys. 
Sorry to bust in like this, but it seemed to be the best place to ask, as you are working on polar kinematics.
i have built a polar printer based on the R-360, using klipper ofc ;)
it uses a printrboard and motors i got from a Q3D oneup i got more or less for free. 
The thing is, i cant seem to find the right number for the bed step_distance.
i got really close it seems, it seemed to be working well before i could use multiple rotations, but now when i try to print a 20x20x20mm cube in each quadrant simultaneously i get a small drift each layer. this makes the cubes slant in one direction. i'm guessing this is because my step_distance is just a little bit off, making each rotation a fraction shorter/longer, thus producing the drift/shift i experience.

the bed is driven by a 1.8 degree stepper @ 1/16 microstepping, one small and one big gear, 10 and 36 teeth, 10 on the motor and 36 on the bed. i tried to calculate the step_distance based on the R-360 marlin configuration and the config example on klippers git, but my math/guessing skills doesnt seem to cut it. 
i was hoping you guys could help me find the sweet spot here. :)   

big thanks in advance :D

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3342987/printer.cfg.txt)
[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/3342988/klippy.log.txt)

KevinOConnor, 2019-06-30:
For your setup, I calculate:

```
$ python
>>> import math
>>> 2. * math.pi * (1.8 / 360.) * (10. / 36.) / 16.
0.0005454153912482281
```

Which is slightly different from the number in your config.

-Kevin

sanpedrox2, 2019-06-30:
thank you very much, you absolute boss! :)

BKLronin, 2019-06-30:
@KevinOConnor 

I see,
I would rather have this fully implemented as well, even more as I have at least a basic understanding of the code right now, that may vanish after some months. 

I thought about that zero offsetting as well but it seemed to dirty to me even if it won´t be realy noticable. You know best whats possible. I could try try the segmentation but that might cost another week. :D

I guess polar.py would be a good spot to implement that?

(Finally got an IDE  that shows some informatio on whats linked etc.)