Klipper3d/klipper #579: CR10S board with CR10 screen

Skraldefar, 2018-08-27:
The firmware works on my printer, pretty surprised actually. Though, i run a CR10s board with a stock CR10 display in it. The screen is simply stuck. Tried both example display configurations, none of them work. On Marlin i know i have to use the CR10S_CR10SCREEN or something, dont know what to do here though :)

KevinOConnor, 2018-08-30:
You'll need to attach the Klipper log file so that we can see what you tried - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

If you set a particular configuration in Marlin then please be specific what that configuration was.

-Kevin

Skraldefar, 2018-08-30:
Hi again. The correct configuration is called #define CR10_STOCKDISPLAY

The log will be attached, sorry, new to this :)
[klippy.log](https://github.com/KevinOConnor/klipper/files/2336693/klippy.log)

KevinOConnor, 2018-08-30:
Does it work if you use:
```
[display]
lcd_type = st7920
cs_pin = ar27
sclk_pin = ar25
sid_pin = ar29
```

Skraldefar, 2018-08-30:
It does. Thank you :)

closed 2018-09-06 by KevinOConnor

Skraldefar, 2018-10-26:
Hi again. I´m trying to use the built in menu. I got the menu to appear by pressing the button with clickpin: ^ar35

Cannot get the encoder_pins working with encoder_pins: ^ar31, ^ar33
I cannot seem to find any information anywhere on what pins i should actually be using. Can anyone help me? I am using a CR10S board with normal CR10 display. Current config is:

`# This file contains pin mappings for the 2017 Creality CR-10S. To use
# this config, the firmware should be compiled for the AVR atmega2560.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
position_endstop: 0.5
position_max: 400

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: pid
pid_Kp: 690.34
pid_Ki: 111.47
pid_Kd: 1068.83
min_temp: 0
max_temp: 130

[fan]
pin: ar9

[mcu]
serial: /dev/ttyUSB0
pin_map: arduino
baud: 115200

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type = st7920
cs_pin = ar27
sclk_pin = ar25
sid_pin = ar29
click_pin: ^!ar35
`