Klipper3d/klipper #581: Lost communication with MCU 'mcu'

dushyantahuja, 2018-08-27:
I'm not sure if I can call this an issue or feature request. But let me explain nevertheless. My 3D printer runs off an ATX power supply (like many others), and I use the raspberry pi to switch on and off the printer.
I switch off the printer whenever it's not printing, however, the pi remains on at all times. When I switch on the printer, klipper gives the following error:

Lost communication with MCU 'mcu'

And needs to be restarted. Many times the restart requires me to switch off the printer a couple of times before klipper can connect back to the printer. It's not too big a problem, but get inconvenient. 

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2325778/klippy.log.zip)

CHILLYSMOKES, 2018-08-28:
Dun worry guys. I got this!! LOL.

Issue the following command in terminal after MCU gets power:
FIRMWARE_RESTART
...then wait about 4-5 seconds.

Or this system command:
echo FIRMWARE_RESTART > /tmp/printer

It clears everything, restarts the MCU(s), and host. Please see the following CLOSED ISSUE for reference:
SYSTEM COMMAND FOR "FIRMWARE_RESTART" AND "RESTART" PLEASE #546

Also, get the Klipper plugin for OctoPrint.
Also, get the PSU plugin so when you turn your printer on it automatically sends that command.

Cheers

blaskkaffe, 2018-08-28:
You could use the control board to switch on and off the PSU, if it is a ramps based one it should have PS_ON pins you can connect the the PSU's green PS_ON wire and purple +5VSB. Then both the control board and the pi would always be on. But the other electronics (steppers and heaters) would be turned off and on.

sameSEA, 2018-08-29:
I also have hard time to turn on printer and get it connected:
STATUS command returns "Printer not ready".
klippy.log has many "Timeout on serial connect"
The port in mcu is according doc "where is my serial port"

 It needs several restarts, firmware resets. 
System command "echo FIRMWARE_RESTART > /tmp/printer" won't help

dushyantahuja, 2018-08-29:
Thanks chillysmokes - I'm already using octo-klipper and PSU-ON -- however I cannot see any option that sends the firmware restart command automatically on switching on the power supply. Can you please help.

CHILLYSMOKES, 2018-08-29:
PSU CONTROL is the plugin, sorry. It gives you the lightning bolt button up top. When you have clicked it and all necessary commands are in place it will only light up when it is done issuing those commands.Take a look at that other issue thread I mentioned above for details. Enter the system commands and check USE GCODE, enter the GCODEs you wanna use for ON/OFF, I used G7 /G8, modify your STOP gcode in Octoprint or SLIC3R to include that GCODE.Note: don't try to use a gcode to turn the PSU on when you hit the PRINT button. It don't work and wigs out Octoprint with errors all over the screen. On, Print.... Done.

dushyantahuja, 2018-08-29:
Great - set my On System command to:
gpio mode 29 out && gpio write 29 1 && sleep 3 && echo FIRMWARE_RESTART > /tmp/printer

closed 2018-08-29 by dushyantahuja