Klipper3d/klipper #583: Feature Request: E-Stop should skip the printer queue and stop immediately

dushyantahuja, 2018-08-29:
E-Stop should skip the printer queue and stop the printing immediately. I think this would be easier on Klipper than on Marlin. Though, not sure how right I am.

KevinOConnor, 2018-08-30:
Klipper does handle M112 requests as soon as they arrive.  If you're having an issue with this, you should attach the Klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

dushyantahuja, 2018-08-30:
Please see attached.
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2338111/klippy.log.zip)

KevinOConnor, 2018-08-31:
I don't see any M112 issued in that log.  The procedure for issuing the M112 is described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md#verify-m112

Run through that procedure and then forward the log from that.
-Kevin

dushyantahuja, 2018-09-01:
Closing this ticket - M112 was not getting issued correctly in my instance. Octoprint was waiting for the print to get cancelled before issuing the M112.

closed 2018-09-01 by dushyantahuja