Klipper3d/klipper #584: Delta config: Tower offset?

balaji-ramani, 2018-08-29:
I have a delta printer that I would like to move to klipper, but the towers have an offset and I don't find a configuration that allows for that.

blaskkaffe, 2018-08-29:
Offset like in diferent distances from the middle or from each other?

balaji-ramani, 2018-08-30:
I am referring to the delta radius of each of the tower being slightly different. In smoothie it is set with M665 A B C settings.

KevinOConnor, 2018-08-30:
The delta radius, arm lengths, and endstop positions can all be individually configured.  An offset is implemented by adding the value in and then configuring the printer with the absolute value.  See the config/example-delta.cfg file the details.

-Kevin

balaji-ramani, 2018-08-31:
I looked at config/example-delta.cfg and the code.  I see that arm length and endstop position can be set individually, but there is just one config value for radius.  Am I missing something?

KevinOConnor, 2018-08-31:
There is only one radius.  (The radius and the tower angles define where the towers are relative to each other.)  If you've got a tower offset of some kind it is likely an endstop position offset.

It would help if you could include the log file of what you've tried in Klipper along with the explicitly smoothie configuration that you use.

-Kevin

blaskkaffe, 2018-08-31:
A picture or info of in what whay they are offset would be helpful

balaji-ramani, 2018-08-31:
This page has a diagram that describes the issue better - http://forum.seemecnc.com/viewtopic.php?t=8698&start=25#p72862.

In the above diagram, due to a build issue the distance of the A, B and C towers on my printer are not the same.  Having the capability to adjust for that in the firmware has allowed me to print dimensionally correct parts.

KevinOConnor, 2018-08-31:
Klipper doesn't support that.  It's just a (confusing) way of specifying the tower angles - changing the distance of one tower is equivalent to changing the angles of the other two towers.

-Kevin

frankvdh, 2018-08-31:
Changing a tower's position is the same mathematically as changing the
printer radius plus changing the towers' radial positions.

If one or more towers is moved from its ideal position (120 degree angle
from other towers and same radius from the ideal centre) in any direction
(including radially), the 3 tower points describe a new circle with a
different radius and centre from the ideal. The tower points are not 120
degrees apart relative to the new centre. So, if you can correct for radius
and tower angle, you can correct for any tower position error (unless the 3
towers are in a straight line ;) ).

The Delta calibrator at escher3d.com allows you to calculate the tower
angles and radius.

I don't know much about CNC, but I guess that the absolute position of the
centre is possibly important there? But for 3D printing, it's not really an
issue.

Frank

balaji-ramani, 2018-09-01:
Thank you for the explanation, Frank.  That cleared it up for me.  I will close this.

closed 2018-09-01 by balaji-ramani

KevinOConnor, 2018-09-06:
FYI, there is an enhanced calibration mechanism for Klipper that is being tested.  This enhanced mechanism supports calibration that take into account XY part dimensions.  See #615 for details.

-Kevin