Klipper3d/klipper #586: menu command "realtime" attribute problem

mcmatrix, 2018-08-29:
```ini
# This is oscillating when doing long moves
[menu __control __move_10mm __axis_x]
type: input
name: "X:{0:05.1f} "
parameter: toolhead.xpos
input_min: -41
input_max: 250.0
input_step: 10.0
realtime: true
gcode: G1 X{0:.1f}
```
Thanks to user @bruce356, who reported about this problem.
> On all axis (XYZ) when `realtime` is set to "true" and one makes larger moves say Z axis after homing is resting at 5mm and I want to raise it to say 170mm, while rotating the rotary encoder and setting the distance to 170mm, Z axis starts to rise but after some distance the axis begins to oscillate randomly up and down, this up and down may continue several times before the Z axis will eventually reach the setpoint of 170mm. Small movements are not affected by this.

@Arksine 
> Also, I have a theory on why the menu's realtime mode could cause oscillation. The menu initializes the input value using the toolhead class. If the user has a gcode transform active this value won't reflect the actual toolhead position while editing, as the input value isn't reset until the user has finished editing. When I get a chance this evening I'll test this theory to see what happens.

I am looking forward to the results of your test.

KevinOConnor, 2018-08-30:
Good catch!  We could queue the messages in gcode.py, but is the menu code itself re-entrant safe?  That is, will the menu code properly handle the case where a call to run_script() pauses and another button event occurs during that pause.  I looked briefly and I suspect that it may not be.  In that case, I think we should build a queuing mechanism (eg, queue_script() ) which never pauses so as to avoid any reentrant issues.

-Kevin

Arksine, 2018-08-30:
Hi Kevin!  I believe you are correct, there is nothing in the menu code itself that handles semi-concurrent calls to run_script.  So create queue_script() function to queue the script, then perhaps have a timer periodically check the queue to see if there is something in it?

KevinOConnor, 2018-08-30:
Yes.  I was thinking something like the below (totally untested):
```
--- a/klippy/extras/display/menu.py
+++ b/klippy/extras/display/menu.py
@@ -512,7 +512,7 @@ class MenuInput(MenuCommand):
         return self._input_value is not None
 
     def _onchange(self):
-        self._manager.run_script(self.get_gcode())
+        self._manager.queue_gcode(self.get_gcode())
 
     def init_value(self):
         self._input_value = None
@@ -901,6 +901,7 @@ class MenuManager:
         self.lcd_chip = lcd_chip
         self.printer = config.get_printer()
         self.gcode = self.printer.lookup_object('gcode')
+        self.gcode_queue = []
         self.parameters = {}
         self.objs = {}
         self.root = None
@@ -1111,8 +1112,8 @@ class MenuManager:
             raise error("Wrong type, expected MenuContainer")
         top = self.stack_peek()
         if top is not None:
-            self.run_script(top.get_leave_gcode())
-        self.run_script(container.get_enter_gcode())
+            self.queue_gcode(top.get_leave_gcode())
+        self.queue_gcode(container.get_enter_gcode())
         if not container.is_editing():
             container.update_items()
         self.menustack.append(container)
@@ -1129,10 +1130,10 @@ class MenuManager:
                     raise error("Wrong type, expected MenuContainer")
                 if not top.is_editing():
                     top.update_items()
-                self.run_script(container.get_leave_gcode())
-                self.run_script(top.get_enter_gcode())
+                self.queue_gcode(container.get_leave_gcode())
+                self.queue_gcode(top.get_enter_gcode())
             else:
-                self.run_script(container.get_leave_gcode())
+                self.queue_gcode(container.get_leave_gcode())
         return container
 
     def stack_size(self):
@@ -1307,13 +1308,13 @@ class MenuManager:
                 self.selected = 0
             elif isinstance(current, MenuInput):
                 if current.is_editing():
-                    self.run_script(current.get_gcode())
+                    self.queue_gcode(current.get_gcode())
                     current.reset_value()
                 else:
                     current.init_value()
             elif isinstance(current, MenuCommand):
                 current()
-                self.run_script(current.get_gcode())
+                self.queue_gcode(current.get_gcode())
 
     def exit(self, force=False):
         container = self.stack_peek()
@@ -1322,7 +1323,7 @@ class MenuManager:
             if (not force and isinstance(current, (MenuInput, MenuGroup))
                     and current.is_editing()):
                 return
-            self.run_script(container.get_leave_gcode())
+            self.queue_gcode(container.get_leave_gcode())
             self.running = False
 
     def run_action(self, action, *args):
@@ -1339,12 +1340,21 @@ class MenuManager:
         except Exception:
             logging.exception("Malformed action call")
 
-    def run_script(self, script):
-        if script is not None:
+    def queue_gcode(self, script):
+        if script is None:
+            return
+        if not self.gcode_queue:
+            reactor = self.printer.get_reactor()
+            reactor.register_callback(self.dispatch_gcode)
+        self.gcode_queue.append(script)
+    def dispatch_gcode(self, eventtime):
+        while self.gcode_queue:
+            script = self.gcode_queue[0]
             try:
                 self.gcode.run_script(script)
             except Exception:
                 logging.exception("Script running error")
+            self.gcode_queue.pop(0)
 
     def add_menuitem(self, name, menu):
         if name in self.menuitems:
```

KevinOConnor, 2018-08-30:
Also, somewhat unrelated, if the menu g-code issues a G1 it probably should always issue an M82 first.

-Kevin

Arksine, 2018-08-30:
That looks good to me.  I'll apply the patch and test it, then report back.  I think the menu can handle multiple gcodes in the config, so I'll add M82 to it as well.

oderwat, 2018-08-30:
@KevinOConnor May I ask why a M82 is needed? Isn't this extruder absolute mode and not related to XYZ axis? While if one wants to extrude by menu it may be better to use relative mode anyway?

Arksine, 2018-08-30:
Patch tested and solves the issue.  Didn't notice any adverse affects.

@oderwat Travels should probably use G90.  With the way the menu is set up, you are right that it should use relative extrusion via M83.

KevinOConnor, 2018-08-31:
Sorry - I meant G90.  Darn g-codes are hard to keep track of.

-Kevin

mcmatrix, 2018-08-31:
Thanks, guys. Nice work.
Please do whatever fixes are needed for this.
As I mentioned in another comment that I don't have fully working Klipper printer
therefor I'm not able to perform any test related with real Axis movements.

bruce356, 2018-09-01:
Hi, has this been applied to the Master version, if not where (which file) do I apply this patch.
thanks - bruce

mcmatrix, 2018-09-04:
Thanks!

@bruce356 How realtime items are working now? Did this patch solved oscillation issue?

bruce356, 2018-09-05:
@mcmatrix @Arksine Hi, thanks for fixing the "realtime: true" for axis XYZ.
I have found an erratic movement problem using "Babystepping" with realtime set to true.
It is very obvious if the encoder is rotated to the maximum limits as set below to say 0.254, the Z axis on my printer moves very rapidly up by several millimetres.
If realtime is not active Babystepping works normally as expected.
 
### menu babystep_adj ###
[menu babystep_adj]
type: input
enable: true
name: "BabySt Z:{0:05.3f}"
parameter: 0
input_min: -0.254	;-0.010" thou of an inch
input_max: 0.254	;+0.010" thou of an inch
input_step: 0.00508	;0.00508mm = 0.0002" thou of an inch
realtime: true 	; true currently not working correctly
gcode: SET_GCODE_OFFSET Z_ADJUST={0:.3f}

Thanks and regards - bruce

mcmatrix, 2018-09-06:
@bruce356 
Hi, correct me if I'm wrong but using real-time mode with the relative adjustment is not the correct way.

>  If the X_ADJUST style parameters are used, then the adjustment will be added to any existing offset (eg, "SET_GCODE_OFFSET Z=-0.2" followed by "SET_GCODE_OFFSET Z_ADJUST=0.3" would result in a total Z offset of 0.1).

It will add that amount to existing offset not setting absolute offset value.
In real-time mode every change in value will execute gcode.
```
Dummy example step: 1, min: 0, max 5. You start from 0 and will scroll to 5.
The executed gcode sequence will be:
SET_GCODE_OFFSET Z_ADJUST=1
SET_GCODE_OFFSET Z_ADJUST=2
SET_GCODE_OFFSET Z_ADJUST=3
SET_GCODE_OFFSET Z_ADJUST=4
SET_GCODE_OFFSET Z_ADJUST=5
and exit from edit mode will execute it again with last value.
SET_GCODE_OFFSET Z_ADJUST=5
So the final offset will be [existing offset]+1+2+3+4+5+5
```

You have to use absolute offset setting.
`SET_GCODE_OFFSET Z=0.2`
The problem is that currently theres no parameter for existing offset.
I have plan to extend gcode module parameter output with home and offset values.
Then it'll be easier to set absolute offset with menu.

bruce356, 2018-09-06:
@mcmatrix hi, thanks for the extensive explanation, it explains why the Z axis moved several millimetres.
I remember reading that section some time ago but had forgotten it.
Have changed from   SET_GCODE_OFFSET Z_ADJUST={0:.3f} to SET_GCODE_OFFSET Z={0:.3f} and now realtime: true works perfectly.
Thank you very much
regards - bruce

bruce356, 2018-09-13:
@mcmatrix hi, Realtime with the extruder is a problem  because for some reason each movement amount adds to the previous amount, example, encoder is set to 10mm movement per notch, if encoder is rotated one notch(+) movement is 10mm (GLCD display shows 10mm), rotate another notch(+) (GLCD display shows 20mm) and movement amount is then  another 20mm (so a total so far of 30mm movement). Instead of each notch moving the extruder by 10mm it seems to move the extruder by the total amount shown on the GLCD display.

Also if the encoder is then pressed to leave this menu option the extruder will then rotate a further 20mm (or whatever amount is displayed on the GLCD at time of pressing the encoder). In this example that means the extruder has moved a total of 50mm instead of only 20mm.

If encoder is rotated one notch(+) movement is 10mm, then rotate encoder back to 0mm  there is no movement. Rotate encoder one notch(-) movement is 10mm in reverse, then rotate encoder back to 0mm there is no movement.

With realtime active but without M83 or G91 there is also unexpected behavour. The movements and the amount of movement per notch is normal and as expected. 
The problem is that if the encoder is pressed to leave the Move E menu option the display returns to 0 and if the Move E menu option is re-selected the extruder returns to its original 0 position instead of remaining at its current position.

Explanation or reply from Arksine:-
The logic for realtime mode seems to be designed with absolute coordinates in mind.   That means the parameter must be toolhead.epos, if it is set to 0 it wont correctly represent the extruder position.    Regardless, I don't think realtime mode is well suited for extrusion.   Relative extrusion without realtime active best achieves the correct functionality.

Is there any way to program/set the E movement to Relative coordinates so that it functions normally in Relative mode using the Realtime option.

Hope this description/explanation is understandable and not too long.

Many thanks again for the great work, regards - bruce

mcmatrix, 2018-09-17:
@bruce356 Hi, could you post your menu config for that extruder movement.
So that I can replicate that action.
Or it's the same as in std menu only with `realtime: true` ?
```ini
[menu __control __move_10mm __axis_e]
type: input
enable: !toolhead.is_printing
name: "Move E:{0:+06.1f}"
parameter: 0
input_min: -250.0
input_max: 250.0
input_step: 10.0
gcode: G1 E{0:.1f} F240
```

bruce356, 2018-09-17:
@mcmatrix hi I have used the 10mm move as an example, behaviour is the same for 1mm and 0.01mm. 

[menu __control __move_10mm __axis_e]
type: input
enable: !toolhead.is_printing
name: "Move E:{0:+06.1f}"
parameter: 0
input_min: -250.0
input_max: 250.0
input_step: 10.0
#realtime: true	;Extruder not function normally with Realtime
gcode: 
      M83
      G1 E{0:.1f} F240

[menu __control __move_1mm __axis_e]
type: input
enable: !toolhead.is_printing
name: "Move E:{0:+06.1f}"
parameter: 0
input_min: -100.0
input_max: 100.0
input_step: 1.0
#realtime: true	;Extruder not function normally with Realtime
gcode: 
      M83
      G1 E{0:.1f} F240

[menu __control __move_01mm __axis_e]
type: input
enable: !toolhead.is_printing
name: "Move E:{0:+06.1f}"
parameter: 0
input_min: -50.0
input_max: 50.0
input_step: 0.1
#realtime: true	;Extruder not function normally with Realtime
gcode: 
      M83
      G1 E{0:.1f} F240

Thanks - bruce

mcmatrix, 2018-09-17:
@bruce356 Thanks.
Btw you can select your code/config part and use `Code` markup.
[image]

bruce356, 2018-09-20:
@mcmatrix hi, on an IDEX printer could you give me some guidance on how to go about setting up menu options for movement that would include the X2 axis and the second extruder (E1).

I have tried 
`[menu __control __move_10mm __axis_x2]
type: input
name: "X2:{0:05.1f} "
parameter: toolhead.xpos
input_min: 0
input_max: 291.0
input_step: 10.0
realtime: true	
gcode: 
    T1
    G1 X2{0:.1f}
`
but this does not work, X still has influence over X2, can it be done or would you have change your program.

Thanks and regards - bruce

mcmatrix, 2018-09-20:
@bruce356 Hi, I assume that you have `[dual_carriage]` config correctly set up 
and both extruders have correct gcode sequence in `activate_gcode / deactivate_gcode` attribute

Can you move those carriages directly from the terminal console?
The menu system itself doesn't have any fancy functionality for this.

I'm wondering how `toolhead.xpos` will do. Does the toolhead have additional position variable for second carriage or the same variable will show second carriage value after `SET_DUAL_CARRIAGE CARRIAGE=1`?

bruce356, 2018-09-20:
@mcmatrix , I can move carriages directly in terminal by selecting the tool (T0 or T1) and then say G1 X125 or any other position.
Have attached my config files
[Klipper printer.cfg & menu.cfg.zip](https://github.com/KevinOConnor/klipper/files/2400998/Klipper.printer.cfg.menu.cfg.zip)
thanks - bruce

KevinOConnor, 2018-09-20:
@mcmatrix - the toolhead shows the position of the active toolhead.  So, after switching to the second carriage, the toolhead will report the position of that second carriage.

-Kevin

mcmatrix, 2018-09-20:
@bruce356 Btw you can now have babystepping in realtime with exact values.
There's new `gcode.homing_zpos` parameters available. 
See the new `Tune` menu item `Offset Z`, visible only in printing time.
```ini
[menu __tune __offsetz]
type: input
name: "Offset Z:{0:05.3f} "
parameter: gcode.homing_zpos
input_min: -5
input_max: 5
input_step: 0.005
realtime: true
gcode: SET_GCODE_OFFSET Z={0:.3f}
```

mcmatrix, 2018-09-20:
@bruce356 
```ini
...
gcode:
T1
G1 X2{0:.1f}
```
Does it has to be `G1 X2{0:.1f}`?
I think it should be normal `G1 X{0:.1f}`
There's no X2 parameter in G1 command.

If that doesn't help then try to remove tool change `T1` command from gcode.
Then try to change tool by using terminal `T1` or `T0` and then try to move X axis.

bruce356, 2018-09-21:
@mcmatrix Hi, thanks for the heads up on the new "tune"menu option. this is a great addition/improvement.
I have used the default menu.cfg for testing, downloaded the latest commit_1747542. The option to set "speed" works , but "flow" and "offset" have a problem. See attached photo.
[image]
Its probably something simple that you would have an answer for.
Regards - bruce

mcmatrix, 2018-09-21:
@bruce356 Are you sure that your code is up-to-date with the master branch? 
This `Tune` menu is using **new parameters** from the `gcode.py` module.

bruce356, 2018-09-21:
@mcmatrix hi, you are right there is no X2 parameter I did not think of that, not sure if what I have done is the best way of going about this, perhaps you could improve this.
``### menu move 10mm ###
[menu __control __move_10mm]
type: list
enable: !toolhead.is_printing
name: Move 10mm
items:
    .__tool_0
    .__tool_1
    .__axis_z
    .__axis_x, .__axis_y
    .__axis_e

[menu __control __move_10mm __tool_0]
type: command
name: "Choose: X_T0 "
gcode: T0 

[menu __control __move_10mm __tool_1]
type: command
name: "Choose: X2_T1 "
gcode: T1 
    
[menu __control __move_10mm __axis_x]
type: input
name: "X:{0:05.1f} "
parameter: toolhead.xpos
input_min: -41
input_max: 291.0
input_step: 10.0
realtime: true		;true	
gcode: G1 X{0:.1f}``

mcmatrix, 2018-09-21:
@bruce356 If it works then at the moment it's the best way.
Currently, the menu is not able to determine which tool is active.

bruce356, 2018-09-21:
@mcmatrix , it works fine and on tool change the location of that tool is displayed on every tool change so I am happy.
regards - bruce