Klipper3d/klipper #588: Feature Request (Menu): Temperature Start

cfelicio, 2018-08-29:
Currently, the temperature start a 0 for both hot end and bed. In other firmwares, it usually starts at 170 for hotend, and 50 for bed. That saves a LOT of wheel turning :-)

mcmatrix, 2018-09-16:
If I'll change temp menu items `input_min:` from `0` to `170` then it won't be possible anymore to change temp below `170`. I have to check it's done in Marlin or in Repetier.

mcmatrix, 2018-09-17:
Or maybe add new feature to menu input. 
```ini
input_step: 1
input_step2: 10
```
So that it'll check scrolling speed and when scrolling fast it will use `input_step2` value.
Normal (slower) scrolling will use `input_step` value.
I'll think about it.

mcmatrix, 2018-09-24:
@cfelicio This fast step rate is now in master.
Please update your codebase and it's should make scrolling faster.
Temp items in the menu now have fast step 10.
So if you scroll faster than 0.03s then inc/dec step will jump to 10.