Klipper3d/klipper #589: Feature Request (Menu): Octoprint Files

cfelicio, 2018-08-29:
So, we have an octoprint menu, which is awesome (and an improvement compared to other firmwares). I'm wondering if besides pausing / stoping a job, it would be possible to read files currently stored on Octoprint, and having the option to start a print on them?

mcmatrix, 2018-08-30:
Isn't that what virtual sd-card does?
```
# A virtual sdcard may be useful if the host machine is not fast
# enough to run OctoPrint well. It allows the Klipper host software to
# directly print gcode files stored in a directory on the host using
# standard sdcard G-Code commands (eg, M24).
#[virtual_sdcard]
#path: ~/.octoprint/uploads/
#   The path of the local directory on the host machine to look for
#   g-code files. This is a read-only directory (sdcard file writes
#   are not supported). One may point this to OctoPrint's upload
#   directory (generally ~/.octoprint/uploads/ ). This parameter must
#   be provided.
```
Currently it's not supporting folders.

cfelicio, 2018-08-31:
Thanks, I just tried the virtualsd option. It does allow me to start a print, but I can't control it via Octoprint afterwards, or monitor the status. I assume that's how it works?

mcmatrix, 2018-09-06:
I think this is the way how sd card printing works. 
When printing via SDCard then Klipper itself is feeding gcodes not Octoprint.
So Octoprint is not able to control printing progress.

Maybe there's a host command to tell OctoPrint to start printing file from a folder.
Then Octoprint will lead the orchestra. The default folder is any way Octoprint upload folder.

KevinOConnor, 2018-09-25:
It looks like this question was answered, so I am closing for now.

closed 2018-09-25 by KevinOConnor

trevjonez, 2019-01-12:
you can start virtual sdcard prints via gcode commands as well. should be able to hook that up to octoprint or any other printer host software. https://github.com/KevinOConnor/klipper/blob/e26d1a356708b3702c91a86ff37f3c5c615941c9/docs/G-Codes.md#g-code-sd-card-commands

shubble32, 2019-10-20:
I'm having a difficult time figuring out how to enable virtual sdcard. Can someone explain how to do it please