Klipper3d/klipper #590: Error loading kinematics 'cartesian'

sqrtmo, 2018-08-30:
Hello, I am just trying to run klipper for the first time. I installed it according to this https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md guide. 
I am facing "Error loading kinematics 'cartesian'" error msg.

Is it something that I am missing? 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2335417/klippy.log)

Thanks in advance!

ozligia, 2018-08-30:
From your log
Error: Unable to parse option 'step_distance' in section 'stepper_x'
Config error
Idk, try to put comment on separate line

oderwat, 2018-08-30:
Yes. I ran into that before too. You can't have an end of line comment. That is not supported by Klipper.

sqrtmo, 2018-08-30:
Ahh, OK. Thanks a lot!

ozligia, 2018-08-30:
@sqrtmo, If our comments helped you and everything is working now, please, consider closing the issue. 
Thanks in advance.

closed 2018-09-06 by KevinOConnor