Klipper3d/klipper #592: Feature Request - Software SPI

GhostlyCrowd, 2018-08-30:
Is it possible to get the ability to use software spi rather then hardware on a ramps/mega setup with TMC drivers.  This would help many Marlin converts not have to rewire their setup.

KevinOConnor, 2018-08-31:
It's certainly possible to implement this.  It's not something I plan to work on.  All the popular boards use the standard SPI ports and I can't think of any reason a user would want to use non-standard ports.

-Kevin

zerg32, 2018-09-10:
It could be useful for makerbot mightyboard. It uses soft spi to talk to thermocouple via MAX6675
I'm even thinking about implementing it myself but not exactly sure how to organize the code.
I have a mks sbase(smoothieboard clone), mightyboard(avr) and ramps for testing/development

KevinOConnor, 2018-09-11:
> I'm even thinking about implementing it myself but not exactly sure how to organize the code.

It would require mcu code changes.  You can look at src/lcd_st7920.c for an example bit banging implementation.  I suspect you'd want to hook into src/spicmds.c:spidev_transfer() so that the existing spi commands can be used.

-Kevin

KevinOConnor, 2018-09-25:
I'm closing this for now - I'll reopen if someone is actively working on it.

-Kevin

closed 2018-09-25 by KevinOConnor

pipersw, 2018-09-28:
It usefull on RAMPS too for people with LCD. I use SW SPI with ramps for TMC2130.

LazaroFilm, 2018-09-28:
That’s too bad. Virtual SPI would allow to avoid making some crazy soldering son Ramps 1.4 with the LCD extension.

koppersvette1, 2018-09-30:
Man I hope you can add software spi support in the near future.  I just ordered a FYSETC F6 control board and 6 tmc2130 drivers with hopes of using it with Klipper.

KevinOConnor, 2018-09-30:
As near as I can tell, the FYSETC F6 uses the standard SPI pins, so it's not a problem.

-Kevin

koppersvette1, 2018-10-01:
Thank you for the prompt response @kevinOConnor. I’m currently waiting for the F6 it’s on the slow boat from China.

luxarts, 2018-10-12:
The spi pins on Arduino Due aren't in the same physical pin than in Arduino Mega. Only can be accessed by the 6 pin in the middle of the board. I use a MAX6675 connected to the SPI pins of Mega (D52 SCK, D50 MISO) on Ramps 1.4 with Arduino Due. The SD reader doesn't work too for the same problem.
When send `status` I get `Max6675 : Device ID error`.

KevinOConnor, 2019-02-18:
FYI, software spi should now be implemented.

-Kevin

Geramy, 2019-05-17:
I can't get software SPI to work on the BIQU SKR v1.3 I keep getting ID Device error for the MAX6675. I do notice the pin i set for the software clk is not pulsing it seems to be just output of a HIGH and not toggling.