Klipper3d/klipper #593: Feature Request: Turn off heaters if printer errors out

cfelicio, 2018-08-30:
Hi Kevin,

I left an ocarina printing last night. I set my max Z at 140mm on Klipper, and the final shut off move on the gcode is to lift Z by 10mm and turn off heaters.

What ended up happening is that the lift Z triggered an out of range move on Klipper (as it likely exceeded 140mm height), and Klipper aborted.

I think that's great as it prevents dangerous moves, but my only concern is that the heaters were left on, and on this particular case, the hot end was left touching the part.

Would be possible to modify the behavior (or have a configurable setting that allows us to do so)?

jakep82, 2018-08-30:
Need a log file.  Klipper does shut down the MCU if an error occurs, so the heaters should have been turned off.

cfelicio, 2018-08-31:
It does not seem to be the case, on both of my printers :(

Another example, today I got an error on my other printer: "Error on 'N16 T3*14': T must have maximum of 0"

Octoprint disconnected, but the MCU kept the heaters on. See attached klippy.log.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2338759/klippy.log)

cfelicio, 2018-08-31:
On the other hand, issuing an M112 will shut off the heaters (it's also on the klippy log towards the end).

BlackStump, 2018-08-31:
@cfelicio  did you follow the Installation recommendations especially under the heading
Configuring OctoPrint to use Klipper
Navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save".

cfelicio, 2018-08-31:
@BlackStump good question, probably not! Going to check tonight and re-try (in case the option is not enabled). Thanks for pointing this out :)

cfelicio, 2018-09-01:
dumb me, this setting fixed it! closing it! hopefully search will help find this one if someone else runs into the same issue! :-)

closed 2018-09-01 by cfelicio