Klipper3d/klipper #595: DELTA_CALIBRATE uses [probe] speed instead of [delta_calibrate]speed while approaching to print bed

ozligia, 2018-08-30:
Log File attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/2337550/klippy.log)
Just before some updates ago while issuing DELTA_CALIBRATE command print head was moving to home position, and after that started to approach print bed at [delta_calibrate] speed. Then there were probings done at [probe]speed. Currently it is doing approach to print bed at [probe]speed, not [delta_calibrate]speed.

Arksine, 2018-08-30:
Ah, Delta's home to the top, so the initial "lift" actually brings it down, and the lift_speed would be slow.  Thanks, I'll make sure keep Delta's in mind in the future.

ozligia, 2018-08-31:
Tested. Fixed. Thank you.

closed 2018-08-31 by ozligia