Klipper3d/klipper #596: Question about DELTA_CALIBRATE

mcmatrix, 2018-08-31:
I'm preparing myself for switching from Marlin119 to Klipper.
I have Anycubic Kossel Linear Plus and V2 type probe.

In facebook group there's instruction for delta calibration.

> If the checks have passed you can start with DELTA_CALIBRATE for autolevel:
> - this step must be repeated 3-7 times until the values only differs at 2-3 digit
>   (e.g. 299.3xx )
> - low speeds work better for default endstops 
> - insert the new values in printer.cfg and restart klipper service (every time)
> - best is to create a new *.txt for copying the values of Octoprint-Terminal after every 
>    DELTA_CALIBRATE run. This helps to get better overview, what your doing.
> 

Kind an annoying procedure, saving config and restarting between iterations. 
If multiple iterations are needed maybe it's good to have these iterations already builtin into calibration functionality. 
Like in Marlin it does multiple iterations automatically until StdDev is minimum and then gives me calibration values.

KevinOConnor, 2018-08-31:
I don't see any gain to running DELTA_CALIBRATE multiple times.  However, a number of delta printers come with probes that don't work well (in particular, small differences in arm length can result in a tilt to the effector which invalidates some probes).  My recommendation is to run DELTA_CALIBRATE with manual probing - it only takes a couple of minutes to run the paper test on the seven calibration points and I've found it returns excellent results.

Separately, I've been working on an "enhanced delta calibration" procedure (there's some stuff up on the work-delta-calibrate-20180828 branch).  Nothing ready to use yet, but I do hope to make the delta calibration process more smooth in the future.

-Kevin

Hywelmartin, 2018-08-31:
I tried something like @mcmatrix :s post suggested and got less and less difference, I settled at the fourth run... maybe there is something to it.. I use a precision piezo Orion so tilt is less of an issue

micheleamerica, 2018-09-01:
I have a probe that uses the nozzle and a switch to detect the bed but its results are far from reliable and I'd like to try the manual method.
What would be the best way to adjust the Z position for each point, sending G0/G1 commands, in relative mode, or use the menu?
And another point, when trying the menu option I stumbled on a problem. I have my printer configured to allow Z to go to -0.75 but I'm unable to move below 0 using the menu during the DELTA_CALIBRATION. Should I further lower the value of minimum_z_position or there some other parameter at play here?
Thanks
Michele

oderwat, 2018-09-01:
@micheleamerica I am using the Octoprint Klipper Plugin with some macros. The I simply open it up on my mobile phone (see: [here](https://github.com/mmone/OctoprintKlipperPlugin/issues/19#issuecomment-416085184)). I think there could be some really nice menu on the printer with the new realtime move options for this. But I am to lazy to try that out.

micheleamerica, 2018-09-02:
Thanks. In fact I've gone the way of the menu.

I've replaced the simple DELTA_CALIBRATE command with the following:

`
[menu __prepare __delta_calib]
type: list
enable: !toolhead.is_printing
name: Delta calibration
items:
    .__delta_calib_start
    .__delta_calib_adjust_z
    .__delta_calib_next_point

[menu __prepare __delta_calib __delta_calib_start]
type: command
enable: !toolhead.is_printing
name: Start calibration
gcode: DELTA_CALIBRATE

[menu __prepare __delta_calib __delta_calib_adjust_z]
type: input
enable: !toolhead.is_printing
name: "Adj. Z:{0:03.2f} "
parameter: toolhead.zpos
input_min: -5.0
input_max: 5.0
input_step: 0.01
realtime: true
gcode: G1 Z{0:.1f}

[menu __prepare __delta_calib __delta_calib_next_point]
type: command
enable: !toolhead.is_printing
name: Next point
gcode: NEXT
`

```

oderwat, 2018-09-02:
@micheleamerica Very cool. I think you have one slight error in it though: The `gcode: G1 Z{0:.1f}` should be `gcode: G1 Z{0:.2f}` if you have 0.01 steps. I wondered why it was not stepping all the time before I saw it. I changed the steps to 0,05 too and max will get 3 for me too.

Would you please edit your comment and made the code block into ``` (you have only one back tick) so that the spaces in front of the `._delta_calib_xxx` lines are becoming visible? Because if people just copy & paste this cool enhancement, they will run into a parsing error as is.

micheleamerica, 2018-09-02:
Thanks for the tip. I had not quite properly tested the code.
And I've updated the comment.

mcmatrix, 2018-09-14:
I know that manual way is always working :)
But is there a better way of doing this with the probe?

I have probe `z_offset` in config. Let's say `16.0`
I'm running `DELTA_CALIBRATION` and inserting result to config.
After `RESTART` i'll execute `G28` and `G1 Z0` when doing paper test it turns out that
there is still a gap between the nozzle tip and bed. Let's say `0.8mm`.

How can I permanently adjust it? Change probe `z_offset` in config and
run calibration again or is there a better way?
In Marlin, I can adjust delta height accordingly and it's done.

mcmatrix, 2018-09-14:
Maybe adding adjustment to homeoverride will do it permanent.
I have to try it.
```ini
[homing_override]
gcode:
   G28
   SET_GCODE_OFFSET Z=0.6
   G1 X0 Y0 Z207 ; delta safe home
```

KevinOConnor, 2018-09-14:
@mcmatrix - it's not clear to me if you're using the probe or using the manual method.  On my delta printer, using the probe just isn't viable - small differences in arm length cause a tilt to the effector - a different tilt at different XY positions.  So, for example, the probe will trigger when the nozzle is 3.0mm from the bed at one XY coordinate and it will trigger when the nozzle is 2.5mm away at another XY position.  If you have a similar probe, then I don't think there's anything productive that can be done with it.

Of course, you might have a better probe (or extremely precise arms).  In that case, if you're finding an offset after probing then adjust the probe's z_offset accordingly.  If you don't want to rerun DELTA_CALIBRATE, then you can use a SET_GCODE_OFFSET command, or you can manually subtract the offset from all three endstop positions.

If you did DELTA_CALIBRATE with manual probing and then found a huge offset, that indicates something went wrong with the calibration process.  Please attach the log from the session (or better yet, pull the enhanced delta calibration code described in #615 and post the log from that).

-Kevin

mcmatrix, 2018-09-14:
@KevinOConnor Sorry, I didn't mention that I was using a probe.
It's standard Anycubic removable probe V2.
My delta arms are as they came from China probaly not very precise.
[image]

I think I got offset between bed and nozzle because inital probe z_offset was not right
but it also could be affected by not precise delta mechanics.

I should try manual calibration and see how I goes. This enhanced delta calibration branch looks interesting.

Thanks for info, I'll try to find out what will work best for me.

ruevs, 2018-09-18:
(Off topic advice from another Anycubic Kossel user)
@mcmatrix  if you add spacers (similar to the brass rivets that Anycubic provided) on the outside of the ball joints you will be able to get 180mm diameter printing. As provided the ball joint sleeves hit the hex nut heads at some points of a 180mm circle. I just bought a 4mm ID, 5mm OD brass tube from a hardware store and cut pieces from it.
See the first comments here:
https://www.youtube.com/watch?v=Bd7Z4JIQjQM&lc=UgjH_wJPzJB6b3gCoAEC.8T51U_GYLJo8T6hiv2SpJ-

totalitarian, 2019-04-28:
Out of interest, do you need to run save_config after each delta_calibrate. Don't the values from the last delta_calibrate stay in memory? I was planning to add a macro that runs g28/delta_calibrate x7 then a save_config

thesleepydog, 2019-05-29:
> I don't see any gain to running DELTA_CALIBRATE multiple times. However, a number of delta printers come with probes that don't work well (in particular, small differences in arm length can result [...]

Is this enhanced delta calibration still looking like making it into production?

Hywelmartin, 2019-05-29:
It's in the main repro...