Klipper3d/klipper #597: Misbehave [homing_override]

sniperlucian, 2018-08-31:
I just switched from working setup with TMC2100 to TMC2130 with  sensorless homing.  If I manually trigger G28 X or G28 Y homing works as expected. 

If I however introduce an  [homing_override]  klippy completely ignores the G28 Y.  It homes X, moves to X230 and than instantly moves Y050 (without homing, so every try moves Y 5 cm further). I also put G4 P500 to cancel out any dynamic effects.

[homing_override]
gcode:
  G28 X
  G0  X230
  G4  P500
  G28 Y
  G0  Y050
#  G28 Z -- comment out for testing
axes: xy

[klippy_not_homing.log](https://github.com/KevinOConnor/klipper/files/2341865/klippy_not_homing.log)

sniperlucian, 2018-09-01:
Solved. Like almost always just a tiny bit of information was missing.

Klippy swallows G28 commands if the printer is bussy, but still remembering following G0 commands. Insert M400 commands after moves, before the next G28 command solves the issue.

I could close it, since the mystery is solved. But still would like to know if the skipping of the G28 commend (if here is still moving ongoing) is wanted behavior !?!?

KevinOConnor, 2018-09-02:
Klipper will process all the G28 commands.  You can run g-code through the host's batch mode - see docs/Debugging.md for the details - if you want to inspect the commands sent.

My best guess is that sensorless homing on corexy is finicky and extra delays help.  (Specifically, it's possible the X-Y stepper is showing as stalled at the start of the Y homing operation and that results in the homing operation ending without any movement.)

-Kevin

sniperlucian, 2018-09-05:
thx for info.

But than I dont understand why it didnt work, even with an additional break and very low sensitivity level  (high value). 

However, it works now an I close this issue.

closed 2018-09-05 by sniperlucian