Klipper3d/klipper #599: Toolhead, M204 GCode failing without S parameter

CameronLamont, 2018-09-01:
Just now having issues using Slic3r PE 1.41 - apparently the S parameter is no longer standard for Marlin but it was in the past. After testing the same model on 1.40 with Klipper I suddenly have an error basically saying there is no 'S' parameter in the M204 commands it now writes up front.

e.g. from my klippy.log
Error on 'N3 M204 P1500 R1500 T1500*84': missing S

Seems this latest version has the following at the very start now - I can't see a config option for it

M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec

I've found this in my searching slic3r/Slic3r#2745 where the S was retired in lieu of separate retraction, printing and travel speeds. Also https://www.reprap.org/wiki/G-code#M204:_Set_default_acceleration

I've had a quick look at the klipper code in toolhead.py; not sure just yet how to make the 'S' parameter optional and equivalent to 'P' (as per Marlin)

Also to recognise the P, R T variables? Guessing they should be equivalent to the acceleration parameters from printer.cfg

CameronLamont, 2018-09-01:
Not sure how to make this check optional, I'll search through klippy code for a similar example to copy

def cmd_M204(self, params):
        gcode = self.printer.lookup_object('gcode')
        accel = gcode.get_float('S', params, above=0.) # this fails if it's missing, ideally look for P instead
        self.max_accel = min(accel, self.config_max_accel)
        self._calc_junction_deviation()

KevinOConnor, 2018-09-02:
This should be fixed now (commit 165317e3).

-Kevin

trgtylcnky, 2018-12-05:
It is not fixed. I can see the change in my clone but still get "missing S" error. I couldn't find where but it seems there is another code looking for "S" in M204 code.

StefanHamminga, 2019-06-21:
Can this be re-opened? I ran into this running the Linear Advance gcode from Marlin:

http://marlinfw.org/tools/lin_advance/k-factor.html

```gcode
; prepare printing
;
M104 S250 ; set nozzle temperature but do not wait
M190 S85 ; set bed temperature and wait
M109 S250 ; block waiting for nozzle temp
G28 ; home all axes with heated bed
G21 ; set units to millimeters
M204 P200 ; set acceleration
```

The `M204` command is called with only `P`. Since the code actually doesn't care if `P`, `T` or `P` and `T` are provided other than picking a minimum[^1] it seems to be reasonable to do what Marlin does and just pick `P` if that's the only provided.

[^1]: https://github.com/KevinOConnor/klipper/blob/3e741a1f88135a3404e7b3eeb3a778305b199893/klippy/toolhead.py#L472

KevinOConnor, 2019-06-21:
The G-Code support in Klipper only targets "common 3rd party software (eg, OctoPrint, Printrun, Slic3r, Cura, etc.) in their standard configurations" as described at: https://www.klipper3d.org/G-Codes.html

That said, it isn't necessary to report an error here, so I changed Klipper to report a warning (commit f6e9db2d).

Separately, to tune pressure advance see: https://www.klipper3d.org/Pressure_Advance.html

-Kevin