Klipper3d/klipper #601: service klipper status : active(exited)

aeropic, 2018-09-02:
Hi,
First of all, thanks for this nice so cool firmware I would want to install...
I'm running octoprint of a rasp pi3 with success.
I followed the instructions to install klipper on this pi. Everything seems running fine (see the log) but I'm afraid I am messing something !
Octoprint does not see the port "/tmp/printer". I try to ls it and it is not found...
Moreover, when checking the status of klipper service I systematically get an active(exited) status...

I tried to delete all the klipper and klipper-env directories and to reinstall klipper and it always behave the same...
What am I doing wrong ? I have to admit I'm a linux-noob !

Thanks for your help.
Alain

log
===
pi@octopi:~ $ git clone https://github.com/KevinOConnor/klipper
Cloning into 'klipper'...
remote: Counting objects: 8795, done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 8795 (delta 22), reused 28 (delta 18), pack-reused 8750
Receiving objects: 100% (8795/8795), 4.59 MiB | 74.00 KiB/s, done.
Resolving deltas: 100% (6402/6402), done.
pi@octopi:~ $ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
[sudo] password for pi:
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libncurses5-dev' instead of 'libncurses-dev'
binutils-arm-none-eabi is already the newest version (2.27-9+9).
build-essential is already the newest version (12.3).
gcc-arm-none-eabi is already the newest version (15:5.4.1+svn241155-1).
libffi-dev is already the newest version (3.2.1-6).
libncurses5-dev is already the newest version (6.0+20161126-1+deb9u2).
libnewlib-arm-none-eabi is already the newest version (2.4.0.20160527-2).
libusb-dev is already the newest version (2:0.1.12-30).
python-dev is already the newest version (2.7.13-2).
python-virtualenv is already the newest version (15.1.0+ds-1).
stm32flash is already the newest version (0.5-1).
virtualenv is already the newest version (15.1.0+ds-1).
avr-libc is already the newest version (1:2.0.0+Atmel3.6.1-1~rpt1).
avrdude is already the newest version (6.3+r1425-1+rpt1).
binutils-avr is already the newest version (2.26.20160125+Atmel3.6.1-1~rpt1).
gcc-avr is already the newest version (1:5.4.0+Atmel3.6.1-1~rpt1).
0 upgraded, 0 newly installed, 0 to remove and 140 not upgraded.

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Also creating executable in /home/pi/klippy-env/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cffi==1.6.0
Collecting pyserial==3.2.1
  Using cached https://files.pythonhosted.org/packages/c0/a7/25d801d71d2eecbbf0f1142a9d38aa77be71172295259dabb7d91310289c/pyserial-3.2.1-py2.py3-none-any.whl
Collecting greenlet==0.4.10
Collecting pycparser (from cffi==1.6.0)
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.18-py2.py3-none-any.whl
Installing collected packages: pycparser, cffi, pyserial, greenlet
Successfully installed cffi-1.6.0 greenlet-0.4.10 pycparser-2.18 pyserial-3.2.1

###### Installing system start script...

###### Launching Klipper host software...
[ ok ] Restarting klipper (via systemctl): klipper.service.

pi@octopi:~ $ ls /tmp
systemd-private-27f3a62b82ba43d585956fdb80a23286-systemd-timesyncd.service-Txo9WN
pi@octopi:~ $ ls/tmp/
-bash: ls/tmp/: No such file or directory
pi@octopi:~ $ ls /tmp/printer
ls: cannot access '/tmp/printer': No such file or directory
pi@octopi:~ $
pi@octopi:~ $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated; vendor preset: enabled)
   Active: active (exited) since Sun 2018-09-02 19:43:51 UTC; 11min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 1241 ExecStop=/etc/init.d/klipper stop (code=exited, status=0/SUCCESS)
  Process: 1255 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/klipper.service

Sep 02 19:43:51 octopi systemd[1]: Starting LSB: Klipper daemon...
Sep 02 19:43:51 octopi klipper[1255]: Starting klipper: klipper.
Sep 02 19:43:51 octopi systemd[1]: Started LSB: Klipper daemon.
pi@octopi:~ $

oderwat, 2018-09-02:
From what I see here you did not configure the hardware part. Check for the parts with `make menuconfig` from the guide. If you did this, did you also create a `printer.cfg` for your printer hardware? You can check this with `more /home/pi/printer.cfg`.

If all of that has been done you could look into `/tmp/klippy.log`. Using `more /tmp/klippy.log` will show the log and you may find the problem from reading it. You can (should) also upload the log here, if you don't find the problem.

aeropic, 2018-09-02:
Thank you oderwat for your fast answer.
Oh yes I did all this. The firmware compiled fine and was uploaded using avrdude command to the ANET board with success using the /dev/ttyUSB0 port at 115200 bauds. The printer.cfg file has also been copied and  is present in the pi directory...

The ANET LCD is fully black.
I came back to MARLIN, just to be sure the board is not bricked, and reloaded the klippy firmware...

the ls /tmp command does not show any klippy log as I cannot connect to the /tmp/printer from octoprint : 
pi@octopi:~ $ ls /tmp
systemd-private-27f3a62b82ba43d585956fdb80a23286-systemd-timesyncd.service-Txo9WN

aeropic, 2018-09-02:
In octopi, from the web interface, I can see two options in the serial ports : 
- AUTO
- /dev/ttyUSB0

no way to see /tmp/printer even after having added it (/tmp/printer) in the serial connections settings

oderwat, 2018-09-02:
Try:

```
cd ~
source klippy-env/bin/activate
python Klipper/klippy/klippy.py printer.cfg
```

This is doing what normally is done by starting the service and outputs what normally goes to the `/tmp/klippy.log`. Maybe that helps to find out what is going wrong.

oderwat, 2018-09-02:
`/tmp/printer` will only be in Octoprint when Klipper is running, which seems not to be the case.

aeropic, 2018-09-02:
I did the python stuff with this result :+1: pi@octopi:~ $ cd ~
pi@octopi:~ $ source klippy-env/bin/activate
(klippy-env) pi@octopi:~ $ python Klipper/klippy/klippy.py printer.cfg
python: can't open file 'Klipper/klippy/klippy.py': [Errno 2] No such file or directory
(klippy-env) pi@octopi:~ $

oderwat, 2018-09-02:
Oh... sorry. I think you need to use a lower case K for the second line! I have an uncommon install.

aeropic, 2018-09-02:
Better results with lowercases   but no connexion

pi@octopi:~ $ cd ~
pi@octopi:~ $ source klippy-env/bin/activate
(klippy-env) pi@octopi:~ $ python klipper/klippy/klippy.py printer.cfg
INFO:root:Starting Klippy...
INFO:root:Start printer at Sun Sep  2 20:47:35 2018 (1535921255.1 75.9)
INFO:root:Extruder max_extrude_ratio=0.266081
INFO:root:Starting serial connect
WARNING:root:Timeout on serial connect
WARNING:root:Timeout on serial connect
WARNING:root:Timeout on serial connect
WARNING:root:Timeout on serial connect
WARNING:root:Timeout on serial connect
WARNING:root:Timeout on serial connect
WARNING:root:Timeout on serial connect

oderwat, 2018-09-02:
OK. There you go. Did you double check the name of the serial device in the `printer.cfg`? You may also have a non standard baud rate chosen or entered. As flashing works its pretty sure that the cable and connection otherwise works.

aeropic, 2018-09-02:
But now there is a printer under /tmp/ ...

(klippy-env) pi@octopi:~ $ ls
klipper  klippy-env  mjpg-streamer  OctoPrint  oprint  printer.cfg  scripts
(klippy-env) pi@octopi:~ $ ls /tmp
printer  systemd-private-d94a3d57660e490c9189232355c5f7a7-systemd-timesyncd.service-UwLbzJ
(klippy-env) pi@octopi:~ $ ls /tmp/printer
/tmp/printer
(klippy-env) pi@octopi:~ $

and from octoprint I now see the printer but get this when trying to connect:
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"Unexpected error while connecting to serial port: /tmp/printer SerialException: '[Errno 2] could not open port /tmp/printer: [Errno 2] No such file or directory: '/tmp/printer'' @ comm.py:_openSerial:2465 (hook default

aeropic, 2018-09-02:
I will double check the printer.cfg...  I'm not sure there is a serial device ?
Anyway, many thanks for this precious help,thanks to you, I'm no more in the dark !!!!

aeropic, 2018-09-02:
you were right ! The serial is wrong :
 [mcu]
serial: /dev/ttyUSB0

THANKS !!!!!!!!

oderwat, 2018-09-02:
Cool! You could also do:

```
ls /dev/serial/by-id/
```

There is most likely just one line with a longer identifier.

Put this into the config as:

```
serial: /dev/serial/by-id/the_name_you_saw_there
```

This will make it more stable if you ever add other USB devices.

aeropic, 2018-09-02:
Sorry I pressed the wrong button and closed the post !
I will try to fix all this tomorrow and come back with the result.

aeropic, 2018-09-02:
you mean this line : 

usb-1a86_USB2.0-Serial-if00-port0

oderwat, 2018-09-02:
Yes. If you use this and for example add a usb webcam it doesn't matter what goes to what port and klipper will always talkt to the printer.

aeropic, 2018-09-02:
Thank you, I'll try...
But coming back to the config file, I don't remember having modified it and the /dev/ttyUSB0 is from the klipper package...
Shall I replace here in the config file  /dev/ttyUSB0  by /tmp/printer   or even by the line you proposed ?

aeropic, 2018-09-03:
Argh ... After a shutdown/restart of the pi, klipper does not start properly anymore (I have manually issue the set of commands you mentionned above
`cd ~
source klippy-env/bin/activate
python Klipper/klippy/klippy.py printer.cfg`

So I think I have a problem with the klipper installation ? But where ?

aeropic, 2018-09-03:
Always with the python set of commands issued manually, I suceeded to connect to the printer through octopi.
but when I send the status command from octoprint, I get an answer : printer is not ready The klippy host software is attempting to connect...

log
==
hanging monitoring state from "Connecting" to "Operational"Send: N0 M110 N0*125Recv: okSend: M115Recv: ok FIRMWARE_VERSION:v0.6.0-397-ga8d4a7e FIRMWARE_NAME:KlipperSend: M21Recv: ok[...]Send: statusRecv: // Printer is not readyRecv: // The klippy host software is attempting to connect. PleaseRecv: // retry in a few moments.Recv: !! Printer is not readyRecv: ok[...]Send: M104 S193Recv: // Printer is not readyRecv: // The klippy host software is attempting to connect. PleaseRecv: // retry in a few moments.Recv: !! Printer is not readyRecv: ok

KevinOConnor, 2018-09-03:
@aeropic - best guess is that the /etc/default/klipper got corrupted on your machine.  You could try:
```
sudo rm /etc/default/klipper
./klipper/scripts/install-octopi.sh
```
After that, run `sudo reboot` to do a full reboot of the Raspberry Pi and then verify that Klipper starts okay on the next boot.  If it fails to start okay, then I recommend you do a complete reinstall of OctoPi (reflash the sd card from scratch) and rerun the Klipper installation instructions.

If you get to the point that Klipper starts correctly, but you get errors (eg, Printer is not ready) then you'll need to attach the full klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

aeropic, 2018-09-03:
Hi Kevin, thanks for your answer!
I tried and got this :+1: pi@octopi:~ $ sudo rm /etc/default/klipper
[sudo] password for pi:
rm: cannot remove '/etc/default/klipper': No such file or directory
pi@octopi:~ $

KevinOConnor, 2018-09-03:
Something is broken with your Octopi setup - as above, I recommend you do a full reinstall of Octopi.

-Kevin

aeropic, 2018-09-03:
Well, huge progress... Now after having performed the install-octopi.sh, the /tmp/printer is created and octoprint can connect to the printer, but when typing status I get the answer : printer is not ready The klippy host software is attempting to connect...

I will try to rebuild and reflash the firmware and keep you informed

aeropic, 2018-09-03:
I am now at the stage where Klipper starts systematically well,  but I get the error "printer not ready".
I followed the instructions to attach the full klipper log but I cannot find it under winSCP. In fact I do not see the /tmp directory...
While under Putty it is present. I made a cat of the klippy.log and here it is (sorry it's my second step with linux after octopi) : 
 log
===
pi@octopi:/tmp $ cat klippy.log
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-397-ga8d4a7e'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Mon Sep  3 19:09:43 2018 (1536001783.0 918.5)
===== Config file =====
[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC2
position_endstop = -30
position_max = 220
position_min = -30
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC3
position_endstop = -8
position_min = -8
position_max = 220
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = !PB2
enable_pin = !PA5
step_distance = .0025
endstop_pin = ^!PC4
position_endstop = 0.5
position_max = 240
homing_speed = 20

[extruder]
step_pin = PB1
dir_pin = PB0
enable_pin = !PD6
step_distance = .0105
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA7
control = pid
pid_kp = 2.151492
pid_ki = 0.633897
pid_kd = 230.042965
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = PD4
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA6
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = PB4

[mcu]
serial = /dev/ttyUSB0

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 1000
max_z_velocity = 20
max_z_accel = 100

[display]
lcd_type = hd44780
rs_pin = PA3
e_pin = PA2
d4_pin = PD2
d5_pin = PD3
d6_pin = PC0
d7_pin = PC1

=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

aeropic, 2018-09-03:
BTW, I tried too to reflash the firmware. My printer is an ANET A8 with the V1.5 board.
Here is the log from Putty, I've the feeling the flash went well ?

pi@octopi:~/klipper $ avrdude -p atmega1284p -c arduino -b 115200 -P /dev/ttyUSB0 -U out/klipper.elf.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "out/klipper.elf.hex"
avrdude: input file out/klipper.elf.hex auto detected as Intel Hex
avrdude: writing flash (19816 bytes):

Writing | ################################################## | 100% 2.66s

avrdude: 19816 bytes of flash written
avrdude: verifying flash memory against out/klipper.elf.hex:
avrdude: load data flash data from input file out/klipper.elf.hex:
avrdude: input file out/klipper.elf.hex auto detected as Intel Hex
avrdude: input file out/klipper.elf.hex contains 19816 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.21s

avrdude: verifying ...
avrdude: 19816 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

The klippy.log above is with the firmware uploaded this way, even if I get the same result with the make flash FLASH_DEVICE=/dev/ttyUSB0 command.

aeropic, 2018-09-03:
IT'S WORKING !!!!!
I just added the baud rate of 115200 in the config file and bang !

I would suggest you fix the ANET-a8-2017.cfg with something like this (the LCD is included and works fine too)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2346499/printer.cfg.txt)

Many thanks again for your help and for the gret job you do with this firmware. I'm now starting using it trying to install my optical probe, pressure control and pushing the beast to its maximal speed ;-)

KevinOConnor, 2018-09-03:
FYI, you should recompile the micro-controller code to use a baud of 250000, flash it again, and remove the baud line from the printer.cfg file.  The anet a8 works fine with a baud of 250000.

-Kevin

stevenvo, 2020-02-01:
> Try:
> 
> ```
> cd ~
> source klippy-env/bin/activate
> python Klipper/klippy/klippy.py printer.cfg
> ```
> 
> This is doing what normally is done by starting the service and outputs what normally goes to the `/tmp/klippy.log`. Maybe that helps to find out what is going wrong.

this saved my day!