Klipper3d/klipper #603: PID Heatbed does not work

muenznej, 2018-09-03:
Hello!
I finished the installation part of klipper and advanced to the verifications and calibrations.
Everything went fine until literatlly the last point in the manual.

Tuning the PID for the Heater worked well. Tuning the PID for the Heatbed did not result in a suggestion of parameters?! It just finished its (5x ?) cycle and did nothing :(

> Recv: T0:26.0 /0.0 B:52.7 /50.0
Recv: T0:26.0 /0.0 B:52.6 /50.0
Recv: T0:26.0 /0.0 B:52.6 /50.0
Recv: T0:26.0 /0.0 B:52.5 /50.0
Recv: T0:26.0 /0.0 B:52.4 /50.0
Send: M105
Recv: ok T0:200.1 /200.0 B:40.3 /0.0
Send: M105
Recv: ok T0:200.1 /200.0 B:40.3 /0.0
Send: M105
Recv: ok T0:200.1 /200.0 B:40.2 /0.0
Send: M105
Recv: ok T0:200.1 /200.0 B:40.1 /0.0

I have a Creality Ender 3.

Hope u can point me a direction.

kind regards
J

[klippy.log](https://github.com/KevinOConnor/klipper/files/2346389/klippy.log)

KevinOConnor, 2018-09-03:
That's certainly odd.  I don't know why OctoPrint didn't show the results.  The results are in the log file:

Autotune: final: Kp=65.914712 Ki=1.339730 Kd=810.750951

-Kevin

muenznej, 2018-09-04:
thank you very much, looking forward to torture my printer.

closed 2018-09-04 by muenznej