Klipper3d/klipper #606: extending module get_status method?

mcmatrix, 2018-09-04:
@KevinOConnor 
Do I understand it correctly that a module `get_status` method output can be freely extended?

For example, the gcode module outputs currently only `speed_factor` and `busy`, would be nice to extend it with additional info like `GET_POSITION` is outputting (home, base etc). 
It makes possible to use `SET_GCODE_OFFSET` in the menu with absolute coordinates.

For `output_pin` and `servo` module we should add `get_status` method and output `last_value`
Currently, the menu functionality is peeking that value directly.

KevinOConnor, 2018-09-04:
> Do I understand it correctly that a module get_status method output can be freely extended?

That's the intent.

> For output_pin and servo module we should add get_status method and output last_value
Currently, the menu functionality is peeking that value directly.

Yes, that should be corrected then.

-Kevin

mcmatrix, 2018-09-04:
Thanks, just wanted to be sure that I’m on the right track.

closed 2018-09-04 by mcmatrix