Klipper3d/klipper #607: Feedback: Nine point delta calibration

ysoldak, 2018-09-05:
I've been trying new experimental [nine point delta calibration](https://github.com/KevinOConnor/klipper/blob/work-delta-calibrate-20180828/docs/Delta_Calibrate.md#nine-point-delta-calibration) recently and here is my feedback.

- [Calibration object](https://github.com/KevinOConnor/klipper/blob/work-delta-calibrate-20180828/docs/prints/calibrate_size.stl) is great, much easier to check for angles with callipers (measuring distance between outside blocks) than trying to align printed object on some paper sheet with angle guides printed, as seen in other objects.
- Hexagons blocks of the calibration object tend to be printed not exactly accurate on all dimensions and that affects measurements. I experience 0.2-0.3mm differences. Probably only matters for very fine calibration.
- [Python script](https://github.com/KevinOConnor/klipper/blob/work-delta-calibrate-20180828/scripts/delta_test.py) is fine, it works and produces results close to the values I have tuned manually to print flat and dimensionally accurate.
- Endstops are adjusted a bit off, though. Probably has something to do with negative z_offset for bed probe. (I have bed positioned on three mechanical button switches and nozzle has to push the bed down -0.3mm to trigger them.)
- In the final version, I'd propose to add finer control on which parameters to adjust. Like user may be sure about arm lengths and wants to calibrate the tower angles only.

Thank you for working on this, delta calibration is a tricky process and better tools definitely needed in this area.

KevinOConnor, 2018-09-06:
Thanks.  FYI, I've made some progress on the full delta calibration system - see #615.

> Hexagons blocks of the calibration object tend to be printed not exactly accurate on all dimensions and that affects measurements. I experience 0.2-0.3mm differences. Probably only matters for very fine calibration.

Interesting. Do you mean the hexagons are deformed or are you referring to extruder swell?  The code subtracts the width of one pillar from the distance between pillars to avoid skew from extruder swell.  So, a deformation shouldn't impact the calculation unless one pillar is distorted significantly different from the other.

> Endstops are adjusted a bit off, though. Probably has something to do with negative z_offset for bed probe. (I have bed positioned on three mechanical button switches and nozzle has to push the bed down -0.3mm to trigger them.)

The test script didn't take into account z_offset at all - the DELTA_ANALYZE code should have that working.

> In the final version, I'd propose to add finer control on which parameters to adjust. Like user may be sure about arm lengths and wants to calibrate the tower angles only.

It's certainly possible to do that (as you've probably seen, the code doesn't really care).  It can be difficult to choose parameters that will properly converge though.

FWIW, on my printer I get better results if I use the arm lengths proposed by the the calibration code, even though I know they don't match my printer's actual arm lengths.  I think minor difference in arm length are causing an effector tilt, and the calibration code is able to account for some of that tilt by using different arm lengths.

-Kevin

ysoldak, 2018-09-06:
> Do you mean the hexagons are deformed or are you referring to extruder swell? The code subtracts the width of one pillar from the distance between pillars to avoid skew from extruder swell. So, a [...]

Several pillars are distorted in my prints, hard to see with naked eye, but once you measure them, you see couple of them "squashed" in one dimension and "elongated" in another. Each of these distortions adding error of +-0.1-0.2mm. So when you then measure pillar distances these errors sneak in.

I've made a remix of your calibration object with 2mm holes in places of pillars, so I can screw M2 bolts into the printed object and do not worry about that pillar distortions. Tried that and got much better readings. Shall probably upload it to Thingiverse and post link here.

> It's certainly possible to do that (as you've probably seen, the code doesn't really care). It can be difficult to choose parameters that will properly converge though.

Yes, sure, coordinate descent does not care. I mean the final GCODE interface, could be nice to be able control which params we want to touch. Something I'm missing in current DELTA_CALIBRATE command as well.

> FWIW, on my printer I get better results if I use the arm lengths proposed by the the calibration code, even though I know they don't match my printer's actual arm lengths. I think minor difference [...]

Same here. At first, me being purist, tried to keep arm lengths fixed, since i know them correct values. Dialing them different (as returned by the script) do produce better results, though.

KevinOConnor, 2018-09-06:
> Several pillars are distorted in my prints, hard to see with naked eye, but once you measure them, you see couple of them "squashed" in one dimension and "elongated" in another. Each of these distortions adding error of +-0.1-0.2mm. So when you then measure pillar distances these errors sneak in.

I suspect you already know this, but just to be clear, the code doesn't really care about the width of the pillars - it's interested in finding the distance between the two pillars (should be 65mm center to center).  If both pillars are distorted in the same way, then the distortion wont matter.  If one is distorted differently from the other then a small error would be introduced.  If that error was .02mm then it would result in a 0.03% error (.02/65) which shouldn't have a huge impact.  If the distortions are due to incorrect delta parameters then going through two rounds could potentially fix that.

> I've made a remix of your calibration object with 2mm holes in places of pillars, so I can screw M2 bolts into the printed object and do not worry about that pillar distortions. Tried that and got much better readings. Shall probably upload it to Thingiverse and post link here.

Interesting.  So, you place a bolt in the printed object and measure the distance between the bolts?

> Yes, sure, coordinate descent does not care. I mean the final GCODE interface, could be nice to be able control which params we want to touch. Something I'm missing in current DELTA_CALIBRATE command as well.

FYI, the latest code starts the calibration with a `CALIBRATE=extended` parameter.  I choose that so that we could potentially extend it with other options in the future.

-Kevin

ysoldak, 2018-09-06:
>If that error was .02mm

Yes. got it, we are interested in distances between pillar centres.
I have all my measured distances differ from each other max 0.5mm. Calibration object scaled 70%, so I expect correct distances (measured, not center-to-center) to be 51.8mm, and this means I have 1% error I want to fix. Pillars themselves can contribute to up to 0.2mm error each (0.5%) due to distortions in the pillars themselves. Subtracting one pillar width may be not enough to get center-to-center distance correct.
Not sure if me just being _special_ trying fix 1% error.
I'm printing with 0.4 nozzle, 0.33 line width.

> Interesting. So, you place a bolt in the printed object and measure the distance between the bolts?

My object has holes in it, and I place bolts in them and screw down all the way, to eliminate any bolt tilt. Then I measure distances same way as with printed pillars, subtracting bolt head dia to get center-to-center distances.
Small holes can be tricky to print, especially on poorly calibrated bot, so in my object i have 0.4mm solid bottom (two layers basically) that metal bolts have no problem to go through later.

> FYI, the latest code starts the calibration with a CALIBRATE=extended parameter. I choose that so that we could potentially extend it with other options in the future.

Shall try!

ysoldak, 2018-09-06:
@frankvdh 
> Possibly the distortion is quantisation error in the slicer? If the width
of the pillar isn't an exact multiple of the nozzle diameter, I think it
will round up or down to the nearest exact multiple?

Can be, I do not know. The result is pillars are not exactly hex, they distorted and distorted differently. Some of them not at all, some of them elongated in one direction.

But as I've said before, it can be really not worth the trouble if all people want is to have their dimensions more or less correct, not 0.1mm correct. Such precision may as well be not even achievable with some designs, steppers and nozzle widths. Probably, something to mention in the docs then later, how much precision one shall expect from calibration.

ysoldak, 2018-09-06:
Here is advertised remix of the calibration object https://www.thingiverse.com/thing:3087261

KevinOConnor, 2018-09-09:
Thanks.  Lets close this in favor of issue #615.

closed 2018-09-09 by KevinOConnor