Klipper3d/klipper #609: Anet A8 LCD Button Pins

davezero78, 2018-09-05:
Hello, how are the

encoder_pins: ^?, ^?
click_pin: ^!?

 for the display on the anet A8?

oderwat, 2018-09-05:
You don't need the pins for the display itself to work. They are used for the menu. But the ANET A8 does not have an rotary encoder. Just the 4 way and enter buttons. Which are implemented as analog input with different resistors for the different buttons. I don't know if that input type is supported for the menu so far. I wrote part of the Marlin version of those buttons as they are implemented now.

You may want to buy an A6 display which works with the Anet A8 board or get a custom rep-rap display and use an adapter like the one I published some time ago (https://www.thingiverse.com/thing:2103748). There are other wirings possible too.

P.S.: I don't have an A8 here anymore.

mcmatrix, 2018-09-06:
> Which are implemented as analog input with different resistors for the different buttons. I don't know if that input type is supported for the menu so far.

Currently, the menu is only supporting separate buttons connected to pin or encoder using two pins.

aeropic, 2018-09-08:
could we imagine a mod using the ANET display as it is and a separate rotary encoder or 4 buttons matrix wired on the pi GPIO pins ?

KevinOConnor, 2018-09-17:
It doesn't look like anyone has any immediate plans to implement this.  So, I'm going to close this for now.

closed 2018-09-17 by KevinOConnor

KevinOConnor, 2019-05-26:
This support was added to Klipper.

-Kevin