Klipper3d/klipper #610: Crealty Ender 3 unknow command M205 over and over and over

dotps1, 2018-09-05:
first off, great product and thank you very much for providing and supporting it.

I got this up and running a few days ago, and I also installed the octopi plug in,  and I noticed I am getting a constant repeting error:

`Unknown Command: M205`

I peeked at the log and I don't even see that in there
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/2354783/klippy.2.log)

here is from the octoprint klipper plugin:
[image]

thanks for any help!

oderwat, 2018-09-05:
You may want to read: https://github.com/mmone/OctoprintKlipperPlugin/issues/22

dotps1, 2018-09-05:
looks like this releated to the plug in and not klipper, thanks!

closed 2018-09-05 by dotps1