Klipper3d/klipper #612: MKS sBase and RepRapDiscount 128x64 Full Graphic Smart Controller

rtbon, 2018-09-06:
Hello, I'm looking to config RepRapDiscount 128x64 Full Graphic Smart Controller with Klipper.
I use a MKS sBase 1.3. After lot of trying nothing seems to work.
My LCD config from smoothie was :
`panel.lcd                                   reprap_discount_glcd     #
panel.spi_channel                           0                 # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16              # SPI chip select     ; GLCD EXP1 Pin 4
panel.encoder_a_pin                         3.25!^            # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^            # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^            # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31              # Pin for buzzer      ; GLCD EXP1 Pin 1
`
Convert to Klipper its looks like that : 
`[display]
lcd_type:st7920
cs_pin: P0.16 
sclk_pin: P0.14
sid_pin: P0.18
encoder_pins: ^P3.25, ^P3.26
click_pin: ^!P1.30`

I made this to fine these pins
[image]

But the screen stay lighted and nothing appear…
I did something wrong or it's not compatible ?
Thank you !

klHaribo, 2018-09-06:
hi! have you tried to flip the cables on the board? with fliped plugs my working config is:
# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
sid_pin: ar17

KevinOConnor, 2018-09-06:
We need the klipper log file in order to assist - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

rtbon, 2018-09-07:
Thanks for your answer @klHaribo, but I already tried to flip plugs and the screen don't light up.
So @KevinOConnor here my klipper log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/2361019/klippy.log)

KevinOConnor, 2018-09-07:
The log output doesn't look right.  Can you reproduce the problem with an unmodified version of Klipper and post the log from that?

-Kevin

zerg32, 2018-09-09:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2363932/klippy.log)

May be my log will be more useful.

Same issue for me. Works fine with smoothieware but doesnt work with Klipper.

KevinOConnor, 2018-09-09:
I think the mks sbase pinout has a typo in it.  What happens if you use:
```
[display]
lcd_type = st7920
cs_pin = P0.16
sclk_pin = P0.15
sid_pin = P0.18
```

rtbon, 2018-09-10:
Thank you very much, you make my day, it works !!!

zerg32, 2018-09-10:
perfect. also works for me

closed 2018-09-10 by rtbon

KevinOConnor, 2018-09-10:
Thanks.  I added a sample to the generic-smoothieboard.cfg example file (commit b595c3e4).

-Kevin

mabl, 2018-11-06:
If somebody ever runs into the same issue:

I found that the connector pins are reversed on my sbase 1.3. Everything works when forcing each of the two EXT connectors reversely into its respective socket. 

To avoid noise from the piezo speeker, also define it as an output to avoid it from floating as an input.

```
[static_digital_output beeper]
pins: !P1.31
```

bshowket, 2019-01-19:
can anyone tell me which pin is 
sid_pin in RepRapDiscount 128x64 Full Graphic? I can't seems to find anything in pinout of exp1 or exp2.

shaiss, 2019-04-08:
I can confirm this works and have a close to working sbase setup.  We should have an example config for the sbase with at least this quirk noted in there if nothing else