Klipper3d/klipper #613: bltouch x_offset y_offset not used while BED_MESH_CALIBRATE

mgth, 2018-09-06:
[klippy.log]

```
[probe]
pin = ar18
x_offset = -39.0
y_offset = -20.0
z_offset = 0.500
```
I did enter the position of the bltouch relative to the nozzle, but obviously it does not change probing position. I understood from the documentation it should be the case.
(z_offset works as expected)

Arksine, 2018-09-06:
X and Y offset do not change the location of probed points, they are used to map the mesh to the bed.  The documentation seems clear that the generated points refer to nozzle position.

mgth, 2018-09-06:
if that's what it means then "refer to nozzle position" is not clear at all to me.
What I understood was if I put min_point: 20,20 it would probe the location where the nozzle would be when moving to abs 20,20
What you tell me if I understand well is : it will move the nozzle to abs 20,20 knowing that it will probe -19,0 in my case ?
I will check with that in mind. Sorry for the confusion then.

Arksine, 2018-09-06:
If min_point is 20, 20, then it would be as if you entered a G0 X20 Y20 gcode, followed by a probe.  After the probe is complete, bed_mesh will use the offsets to map the mesh to the locations that were actually probed, such as -19, 0.

mgth, 2018-09-06:
that's clear enough then. And it is what I get. Thank you much for the precision.

closed 2018-09-06 by mgth

cristianku, 2019-07-17:
let me understand, if i put a negative x_offset , lets say -50, it means that my BLTOUCH is located on the left of the nozzle ?

is it right ?