Klipper3d/klipper #616: servo probe behaviour with mesh bed leveling

aeropic, 2018-09-06:
First of all, I must say I'm amazed by the features  Klipper now offers. i've installed it on my ANET A8, tweaked the printer config in order to add a servo probe and mesh bed leveling, and every thing is working (I mean first print OK after bed sensing!)

 My question is, is it normal the servo is activated before and after each probe point. I have not found a way to activate the servo once to release the probe, sense all 9 points on the grid, then activate the servo to place  the probe in the parking position.

I think a way to do this could be to add the servo move in the [homing_override] gcode section
and
do the same in the [bed_mesh] gcode  section if it would exist !!!!

Is there another way ?
Thank you for this amazing piece of software.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2358440/printer.cfg.txt)

oderwat, 2018-09-06:
Good idea. This would need a bed_mesh start and end gcode section. I don't see how it would be related to the homing override section. This would only be useful if one takes more than one probe sample I guess.

oderwat, 2018-09-06:
BTW what probe do you use? I recently made this mod for using the BFPTouch which works much better than all three of the BLTouch clones I testet. https://www.thingiverse.com/thing:3080865

aeropic, 2018-09-06:
Hi Oderwat,

I 'm using this optical Z probe which is really very very accurate in terms of repeatability.  [](https://www.thingiverse.com/thing:2805327)
When homing the Z, you need to have the probe activated. Today it is done in the probe gcode section : `activate_gcode:
	SET_SERVO SERVO=bltouch ANGLE=120
	G4 P200
deactivate_gcode:
	SET_SERVO SERVO=bltouch ANGLE=10
	G4 P100`

If you suppress those lines, z-homing will fail as ther would not be any z switch ...

oderwat, 2018-09-06:
That is similar to what I do. I just wanted to know which probe you use. I do not really see a reason for not storing the probe in between points but as I said it could be done if mesh bed gets a start and end gcode section (which should be trivial to implement).

aeropic, 2018-09-06:
Yes, It is basically the same optical endswitch used in our probes ! Indeed far better than BLtouch...
Not storing the probe between points would just allow to probe faster and would also limit the stress on the servo ...

Arksine, 2018-09-06:
Could you remove the activate-deactivate gcodes from the probe and use a gcode_macro?

```
[gcode_macro G80]
gcode:
SET_SERVO SERVO=bltouch ANGLE=120 
G4 P200 
BED_MESH_CALIBRATE
SET_SERVO SERVO=bltouch ANGLE=10 
G4 P100 
```

aeropic, 2018-09-07:
Clever idea I will try soon.
I think it will work for the mesh leveling but I have some doubts for both the homing and leveling.
Indeed when mesh bed leveling, there is a homing which is performed hence it will work. but if we perform a homing without the probe motion in the probe section it will crash into the bed... And if we add a gcode to activate the probe the probe in the oming override section then I'm afraid the mesh leveling will fail as there is systematically a homing which is performed before the probing...
maybe would it be better to suppress the homing in the mesh leveling ?

I will try and keep you informed!

aeropic, 2018-09-08:
Hi friends,
I found the way to patch klipper !!!
Just removed this line in bed_mesh.py : 
def start_calibration(self):
        self.bedmesh.set_mesh(None)
        #self.gcode.run_script_from_command("G28") <<<<<<<<<<<<<<<<<<<<<<<<
        self.probe_helper.start_probe()
[bed_mesh.py.txt](https://github.com/KevinOConnor/klipper/files/2363167/bed_mesh.py.txt)

now the probe section becomes:
# Define a probe using the BLTouch
[probe]
pin: ^PC4
x_offset: -13
Y_offset: 22
z_offset: 2.41
;activate_gcode:
;	SET_SERVO SERVO=bltouch ANGLE=120
;	G4 P200
;deactivate_gcode:
;	SET_SERVO SERVO=bltouch ANGLE=10
;	G4 P100

the homing override:
[homing_override]
set_position_z: 5
axes: z
gcode:
	G90 ; Uncomment these 2 lines to blindly lift the Z 5mm at start
	G1 Z10 F600
	G28 X0 Y0
	G1 X112.5 Y112.5 F3600
	SET_SERVO SERVO=bltouch ANGLE=120 
	G4 P200
	G28 Z0
	SET_SERVO SERVO=bltouch ANGLE=10 
	G4 P100 

and the macro to mesh probe the bed:
[gcode_macro G29]
gcode:
	G28
	G1 Z10 F600
	SET_SERVO SERVO=bltouch ANGLE=120 
	G4 P200 
	G1 Z10 F600
	BED_MESH_CALIBRATE
	SET_SERVO SERVO=bltouch ANGLE=10 
	G4 P100 
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2363168/printer.cfg.txt)

It is now working as with MARLIN and the slicer start code has not to be modified as the G29 macro does the same job !
Klipper is really great and fun, thank you for all the work you do here.

aeropic, 2018-09-08:
Thanks !
I really think the probe activation should be connected not to a single probe but to the "macro" actions that require a probe:
- z homing
- bed calibration
Even if with a servo probe the way it behaves today could be OK, if you use a more weird release system (HallOn, Touch-Mi) that would be a mess. (see https://github.com/KevinOConnor/klipper/issues/620)

Concerning homing inside the probing process or not, I would say you could check and forbid the probing if homing is not done, I think this would make a more versatile sofware ... Homing is homing, probing is probing ;-)

Eric, If you're at the origin of the mesh leveling module: congratulations as it was exactly what I wanted before jumping into Klipper's adventure !
Have you seen the other issue I get with too high Z-speed after leveling? I tried to dive into the code but could not figure out where to search...

aeropic, 2018-10-08:
I close this one as the #751 is fully covering the only open point (G28)

closed 2018-10-08 by aeropic