Klipper3d/klipper #617: How do I setup a multi-step process?

yoyo42, 2018-09-06:
I'm trying to add support for a thermistor that needs two output pins to be enabled, a reading taken, then the pins disabled again. I can't see where in the code to add these sequential steps. I suspect I'll need to add a new thermistor class.

I can't just static enable the pins because the thermistor (a PT100) heats itself up and gets inaccurate if you leave it powered up.

Where should I add this code? Do I just need to schedule the steps separately at specific times? Hopefully not...

yoyo42, 2018-09-08:
Great, thanks. I'll make a pull request it if I get it going well - it will add support for the Cetus with Tinyfab replacement CPU (==Smoothieboard).

KevinOConnor, 2018-09-17:
Okay - I'm going to close this for now.  Feel free to open a pull request when ready.

-Kevin

closed 2018-09-17 by KevinOConnor