Klipper3d/klipper #618: Feature request - XYZ probing

coldelectrons, 2018-09-07:
My eventual goal is to implement a calibration routine analogous to Prusa's, using inductive sensor points.  One stepping stone there is a more generalized probing function.

The behavior I have in mind is this:  From current position, probe towards a given position, stopping when the probe changes to a desired state.

What are your thoughts?

KevinOConnor, 2018-09-09:
I'm not sure what's being asked.  The current code does move towards a point and it stops when the probe triggers.

-Kevin

KevinOConnor, 2018-09-12:
That is internally how the code works (see the homing.py:homing_move() and extras/probe.py:cmd_PROBE() methods for the details).  Performing this type of action would require host python code to be written.

-Kevin

closed 2018-09-25 by KevinOConnor