Klipper3d/klipper #619: Z hight

fomoco460, 2018-09-07:
for the life of me i can not get my z to move 10mm in octoprint.The most i'v been able to get it to go is 6.22 when I ask it to go 10. Not sure what I'm doing wrong.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2362728/klippy.log)

KevinOConnor, 2018-09-08:
Did you run the checks described in: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

Have you seen the FAQ item: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-converted-my-config-from-marlin-and-the-xy-axes-work-fine-but-i-just-get-a-screeching-noise-when-homing-the-z-axis

If so, please try running the command you are having issue with, then immediately issue an M112 command, and then attach the klipper log - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

fomoco460, 2018-09-08:
I have done all the checks and read the faq. here is my log from running a home then 10mm up.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2363573/klippy.log)

KevinOConnor, 2018-09-08:
The log indicates Klipper correctly scheduled the appropriate steps.  If you're reliably getting smaller movement than requested it likely means your step_distance parameter is not correct.  If you're not getting reliable movement at all, it may be a mechanical or wiring issue.

-Kevin

fomoco460, 2018-09-08:
I'v tried to adjust my stepper_distance but if i go to high with it my motors just buzz. in marlin everything works as expected.

fomoco460, 2018-09-08:
Im going to close this now. I did a fresh install of octopi and klipper and now things seem to be working. adjusting my stepper_distance is now getting really close to 10 just need to dial it in a bit more.

closed 2018-09-08 by fomoco460