Klipper3d/klipper #620: Probe - Activate Gcode (move) not supported

NedalLive, 2018-09-08:
I use a touch-ME as probe. I have to add a Gcode (movement) to deploy it, i put it in activate_CODE of probe.
I add an home_override to first home X and Y, and after a Z home. I added this to fix the home position before the start of gcode deployment.

When i start the G28 :
[klippy.log](https://github.com/KevinOConnor/klipper/files/2362832/klippy.log)

  the printer start the G28 X and Y, all is fine,
  next it move to X100 and Y100 (where i want to probe the bed), it's ok too
  next step is to deploy the probe, and i see an unexpected movement :
    the bed level up to probe, during the movement to the deployment position (the movements mix at very low speed)
    even if the probe fire a trigger event, the G28 Z is not ended.
I think Activate_Gcode does not support motion management to enable the output of mechanical probe

aeropic, 2018-09-08:
Hi,
I'm not sure it work the way you want. As far as I understand, the way the probe works is that the activate/deactivate gcode sections are run everytime the probe as to probe a point on the bed and this will not work with a touch-ME probe.
I have a servo activated probe and made some mods in Klipper to make it work the way I want ;-) dee here : [https://github.com/KevinOConnor/klipper/issues/616](url)

NedalLive, 2018-09-08:
Thank you @aeropic  for your answer,
no need to replay the probe output gcode between each probe.
only at launch.

I countered the problem by waiting for the correction by making macro launch

KevinOConnor, 2018-09-08:
It is not valid to put movement commands (eg, G1, G28) in the probe's activate_gcode and/or deativate_gcode config.  Those fields are intended to enable the probe (eg, setting a pin, or setting a servo).

-Kevin

NedalLive, 2018-09-08:
Hello @KevinOConnor , thank you for additional information.
that's what it seemed to me ...
In the case of probe with mechanical activation would not it be wise to make this option possible?

aeropic, 2018-09-09:
If you would place the mechanical activation at probe level, this would imply, in your case with Touch-me, the head would go to the X endstop to release the pin then go to the probe position, sense one point, go down to park the probe at each probing point of the grid
==> a mess!

KevinOConnor, 2018-09-12:
If a probe needs some kind of physical movement in order to activate then it would require python host code to be written to support it.

-Kevin

closed 2018-09-25 by KevinOConnor