Klipper3d/klipper #622: Internal error on command:"PID_CALIBRATE"

mcmatrix, 2018-09-08:
A strange error was thrown during the heated bed PID calibration.
I'm using Repetier-Server and Raspberry Pi3B+

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2363220/klippy.zip)

KevinOConnor, 2018-09-08:
This looks like a quirk in Repetier Server - it stopped reading data from Klipper to the point that the OS buffer filled up.  Klipper should have handled it better though.

-Kevin

mcmatrix, 2018-09-08:
It felt not normal. 
From UI (Repetier-Server) point of view, the klipper was unresponsive during PID calibration.
Maybe the PID calib. cycle is behaving somehow different.

KevinOConnor, 2018-09-09:
There's nothing different on the Klipper side (other than a longer time to the "ok" message).  You'd need to check the repetier server side for differences.

mcmatrix, 2018-09-09:
Thanks, I’ll play around and see how Repetier-Server will act.

KevinOConnor, 2018-09-13:
FYI, I've updated Klipper so that it wont throw an internal error on a failed write to the pseudo-tty.  (However, the written content is lost, so it's still important that octoprint/repetier server still reliably reads incoming data.)  Commit 91691afd.

-Kevin