Klipper3d/klipper #624: z-probe offset (mesh leveling)

aeropic, 2018-09-08:
Very weird... I get a Z probe which is sensing the bed 2.41 mm under the tip of the head.
I then insert a 
z_offset: 2.41   in the [probe] section

When printing a calibration cube with mesh leveling activated, I get a cube with a strong distortion in Z: it measures 22.40mm while the other dimensions are correct.

it is not a problem of step_distance in z (it's an ANET A8) ==> 0.0025

When starting to print, the LCD gives a -2.1mm but it finishes its last layer at 20.1 mm with 200 layers, the first one at 0.3 the other ones at 0.2mm
==> I get the feeling the z-offset is included inside the computations and the layers are then ~10% too big ...

I de-activated the [bed_mesh] section but kept the probe as a virtual z-switch, with this the G28 is executed and when starting the print the first layer flies 2.4mm above the bed...

Where am I wrong ?

aeropic, 2018-09-08:
Let me self close this issue ;-)
I just forgot to set the position_endstop: 2.41 (same value as the z probe offset)

I must admit that comming from MARLIN, this could be somewhat misleading to have to enter twice the same value. In fact, as the endstop_pin is declared as "probe:z_virtual_endstop" my assumption was that the position_endstop could be left to 0 ...

Maybe be a future release could simplify this ?

closed 2018-09-09 by aeropic

fessyfoo, 2019-02-15:
a future release did solve this: 

https://github.com/KevinOConnor/klipper/commit/c5d4e14298525e0a184f42f812041b32dfea6951#diff-1e43906dac5cad94f5646fcbef9a5d0e