Klipper3d/klipper #627: z endstop doesnt change, is always open Tevo Tornado MKS Base 1.4

Hanks552, 2018-09-09:
Well, im not a expert, probably the printer.cfg is with the wrong value, X and Y works fine... So im wondering if anybody can help me or have the profile for the tevo tornado with MKS_Base 1.4. Thanks

Im using generic-ramps.cfg as a start point, i only changed the inverted endstop in 1 axis and changed the bed size to 300x300x400

J-charles-C, 2018-09-09:
I,

**In your stepper Z :**
[stepper_z]
**endstop_pin: probe:z_virtual_endstop**
#endstop_pin: z_virtual_endstop

**For Bl-Touch :**
# Define the BLTouch servo
[servo bltouch]
pin: Pin of your card for SERVO1 (D11)
maximum_servo_angle: 180
minimum_pulse_width: 0.0006
maximum_pulse_width: 0.0024

# Define a probe using the BLTouch
[probe]
pin:  Pin of your card for Zmin endstop
z_offset: 1.803
activate_gcode:
   SET_SERVO SERVO=bltouch ANGLE=10
   SET_SERVO SERVO=bltouch ANGLE=60
   G4 P200
deactivate_gcode:
   SET_SERVO SERVO=bltouch ANGLE=90
  G4 P200

best

J-charles-C, 2018-09-09:
Yes it servo1 pin. D11 or D12

best

Hanks552, 2018-09-09:
Thanks for your reply, but i just dont understand the pinout numbers... because they repeat in other places, thats crazy

closed 2018-09-09 by Hanks552