Klipper3d/klipper #629: BLTouch and Klipper Tevo tornado MKS Base 1.4

Hanks552, 2018-09-09:
i cant make it work... i set it to 4 and everytime i try to print it, the nozzle hit the glass 0.3...
the instructions say "set the endstop_position in the stepper_z section to match the probe's z_offset." but there is not endstop_position, we have position_endstop... (should i creat this command or is just wrong in the instructions?)

[image]

Thanks! My Marlin Z offset was -2.9... if there is any tutorial to setup z offset on Klipper, please send to me, because i only know for Marlin, with the paper, current Z offset + the offset with paper = new offset

KevinOConnor, 2018-09-09:
The comment in the config was a typo - it should have read 'position_endstop' - it is fixed now (commit ea129d86).

In order for us to assist, you must attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Hanks552, 2018-09-09:
one more thing, the nozzle is not in the middle when the bl touch is started, and in marlin it was doing a LOT of leveling before it starts to print, this one here just do 1 time
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2364610/klippy.zip)

thanks for your help, here is the log file

Hanks552, 2018-09-10:
and i notice that the auto bed leveling is just hitting 1 spot, and when i tried bed_tilt_calibrate the BLtouch is outside the bed, so i need to set offset for this...
[image]

KevinOConnor, 2018-09-12:
It's not clear to me what you are asking or if you're reporting a problem.  If it's a problem could you describe what you ran, what you saw, and what you expected to see?

-Kevin

closed 2018-09-25 by KevinOConnor