Klipper3d/klipper #630: Heater extruder not heating at expected rate after FIRMWARE_RESTART

mgth, 2018-09-09:
If I send a firmware restart while hotend is hot. And then ask for 200°C the heater will go up for about 10°C. and then fail with "Heater extruder not heating at expected rate". the rate was actually slower than usual.

But if I wait the temp to reach below 100°C it works ok.
This is very reproducible. 

[image]

mgth, 2018-09-09:
sorry, capture failed
[image]

jakep82, 2018-09-09:
You need to post your log file. You probably just need to adjust the heating_gain and/or check_gain_time settings.

KevinOConnor, 2018-09-17:
As mentioned, one can tune the verify_heater sections to avoid this.  Another option is to not rapidly heat, restart, and reheat.

closed 2018-09-17 by KevinOConnor